Hypfer/Valetudo #1366: WebUI / http is no longer working

domibarton, 2022-01-20:
## Describe the bug

I installed 2022.01.0 on my Roborock S50. It worked flawlessly, even with the Home Assistant / MQTT integration.
Unfortunately, now the WebUI is no longer reachable / working. I can ping the vacuum and ssh into it. I can also reboot the vacuum and it works. However, the WebUI is no longer working (properly).

I get the HTTP 401 response and I can authenticate, so I get a HTTP 200 back. But when the static files are loaded, the files are pending, the WebUI hangs and there's no more response.

Funnily enough, the Home Assistant integration via MQTT works!

## To Reproduce

1. Go to the web URL
2. Enter the credentials
3. Login
4. Wait…

### Screenshots

[image]

The `/static/js/2.b5dc54c8.chunk.js` file is pending. 
When I open the URL directly, it will load endlessly:

[image]

### Vacuum Model

Roborock S50

### Valetudo Version

2022.01.0

## Expected behavior

1. The WebUI and all static files area loaded

## Additional context

Is there any way how I can further debug this issue? What I did after the stock setup:

- Configured HTTP Basic Auth
- Configured MQTT (Host, Port, Credentials, Prefix, Topic Name)
- Activated the MQTT map thingy
- Reduced the volume

What I've tried so far:

Found `valetudo` running:

- SSH'd into the robo and rebooted it (several times)
- Deactivated `mqtt` via `/mnt/data/valetudo_config.json` and rebooted again
- Browsed other issues in this repo to find similar issues

What I found was was [this discussion](https://github.com/Hypfer/Valetudo/discussions/1357).  
What is `_root_postboot.sh`, because this is missing:

```
[root@rockrobo bin]# ls -l /mnt/data
drwx------    2 root     root         16384 Jan  1  1970 lost+found
drwxr-xr-x    2 root     root          4096 Jan 20 00:00 miio
drwxr-xr-x    6 root     root          4096 Jan 20 00:00 rockrobo
drwxr-xr-x    3 root     root          4096 Jan 18 22:16 rriot
-rw-r--r--    1 root     root          1829 Jan 19 23:53 valetudo_config.json
drwxr-xr-x    2 root     root          4096 Jan 18 22:18 wlan
```

However, valetudo is running:

```
[root@rockrobo bin]# ps | grep vale
  365 root      132m S    /usr/local/bin/valetudo
 8098 root      2268 S    grep vale
```

Please note I'm using private CIDR network ranges (not like others ^^). My client is in a different VLAN / subnet than the roborock. There's masquerading between does networks. However, I can also connect the client to the same subnet as the robo and it still doesn't work.

domibarton, 2022-01-20:
If it helps, if I open `top` on the robot, it will freeze and I've to kill my terminal and reconnect.

domibarton, 2022-01-20:
Mhm can't even `scp` the config:

```
scp vacuum-king-arthur.barton.confirm.ch:/mnt/data/valetudo_config.json .
valetudo_config.json                                                                                                                       0%    0     0.0KB/s   --:-- ETA
```

Looks a bit like a WLAN issue on the roborock - or I/O. Any ideas or hints I can further debug would be appreciated. I've A LOT of WLAN devices, but none of them do have those issues :/

closed 2022-01-20 by Hypfer

domibarton, 2022-01-20:
@Hypfer just closing the issue? why is it invalid? I'm just asking for ideas how to debug this even further…

Hypfer, 2022-01-20:
It is clearly not a bug.

See also http://www.catb.org/~esr/faqs/smart-questions#idm368

domibarton, 2022-01-20:
> It is clearly not a bug.
> 
> [image]
> 
> See also http://www.catb.org/~esr/faqs/smart-questions#idm368

I was fairly certain it is a bug, since I reverse engineered & debugged the HTTP call. I get a 401, then a 200 on successful auth, index is transmitted, static file is hanging. It looks like a bug and feels like a bug. I tried to pinpoint it down, but I wasn't successful and just hoped I get some help.

I'm aware of "supporting the FOSS community" can be a PITA and I'm quite active in the community myself as well. In fact, I created the [`python-miio` troubleshooting guide](https://python-miio.readthedocs.io/en/latest/troubleshooting.html) and did a lot of reverse engineering regarding network issues & NAT / masquerading problems. I also try to help other users, [right here for example](https://github.com/rytilahti/python-miio/issues/422#issuecomment-455824152).

 I just thought I might get some help or "directions where to debug" here as well, thus these quotes:

> Is there any way how I can further debug this issue? What I did after the stock setup:

> Any ideas or hints I can further debug would be appreciated.

But nevermind: Your project, your rules - I accept that. I'll continue reverse engineering it myself.

Btw. kudos to such an amazing project. Thank you for your effort :)

Hypfer, 2022-01-20:
> I reverse engineered & debugged the HTTP call

Yes yes. The magic reverse engineering button labelled F12