Hypfer/Valetudo #1373: React Frontend Code Review

Hypfer, 2022-01-23:
If you're a senior react developer looking for something to contribute to the project here's something.
As the new frontend has been my first experience with reactjs, I'm fairly certain that the things I did there do leave some room for improvement. It would be great if someone experienced could take a look at the codebase.

I'd expect some non-best-practices, opportunities to improve performance and similar findings.
Hooks used incorrectly or not at all even if they could be useful. Stuff like that.

What I mostly don't care about is formatting. I'm pretty happy with how the code looks and reads as it is currently.
Also, I'm already opinionated enough so I don't need any additional opinions from e.g. prettier or standardjs.

Here's a meme:
[image]

adminy, 2022-01-23:
One option would be to upgrade to preact. Almost no code difference with boost in performance. Especially since you're not using the likes of redux, its perfect in that case, otherwise you would have to alias `preact/compat`. I switched to preact relatively easily for all my projects.

The concern I have is the OOP, I come from functional background and this to me feels like a jungle, don't take it personally.

Also some files are lengthy, perhaps they need some splitting, I have a columns/rows limiter and its breaking over on a couple of occasions. But I'm sure most of it is just probably a @mui/material artefact, which you seem to know it better than me, I'd probably be on the first part of the meme stepping on it. I haven't moved on from bulma, it has been too good to me. Consider modularising some of it because the repo contains too much code to look at.

```
104 vulnerabilities (93 moderate, 9 high, 2 critical) <---- probably react dependencies.
```

tgrrr, 2022-02-06:
Hi,

Why are you nesting router switches inside one another? Is it an intentional decision?

It may cause some problems with inheriting state.

Cheers

tgrrr, 2022-02-06:
Would you prefer feedback inline with the code, or as a PR?

Hypfer, 2022-02-07:
> Why are you nesting router switches inside one another? Is it an intentional decision?

Would it still be possible to have the router split into multiple sub-routers without doing that?

> Would you prefer feedback inline with the code, or as a PR?

I think a comment here referencing the code in question would be best

closed 2022-11-29 by Hypfer