Hypfer/Valetudo #1423: roborock s5 max: too many forbidden markers to save!

Lanchon, 2022-02-21:
## Describe the bug
cannot save virtual restrictions. not even if i don't make any changes.

## To Reproduce
go to virtual restriction management. make no changes. click SAVE. error pops up:
> An error occurred while sending command to CombinedVirtualRestrictionsCapability: too many forbidden markers to save!

### Vacuum Model
roborock s5 max latest fw (1566, 11/2021) just rooted

### Valetudo Version
2022.02.0 dde8f2ef28d8c7cd6b7456f6628595f9613dbf74

## Expected behavior
being able to save changes.

## Additional context
i currently have 7 zones and 4 walls, amounting to just 36 vertices.
the no-gos were defined using the roborock app on stock 1566 fw.

Hypfer, 2022-02-21:
Please check what is being sent by the frontend when saving by using the browser devtools.
Also, it would be interesting to know, how many entities are there in the map data sent by the API

There was an issue with zones being duplicated before but iirc that was fixed.

Lanchon, 2022-02-21:
```
{
  "virtualWalls": [
    {
      "points": {
        "pA": {
          "x": 3050,
          "y": 2820
        },
        "pB": {
          "x": 3170,
          "y": 2820
        }
      }
    },
    {
      "points": {
        "pA": {
          "x": 3060,
          "y": 2930
        },
        "pB": {
          "x": 3170,
          "y": 2930
        }
      }
    },
    {
      "points": {
        "pA": {
          "x": 2670,
          "y": 2490
        },
        "pB": {
          "x": 2670,
          "y": 2625
        }
      }
    },
    {
      "points": {
        "pA": {
          "x": 2515,
          "y": 2320
        },
        "pB": {
          "x": 2625,
          "y": 2320
        }
      }
    }
  ],
  "restrictedZones": [
    {
      "type": "regular",
      "points": {
        "pA": {
          "x": 2610,
          "y": 2310
        },
        "pB": {
          "x": 2700,
          "y": 2310
        },
        "pC": {
          "x": 2700,
          "y": 2495
        },
        "pD": {
          "x": 2610,
          "y": 2495
        }
      }
    },
    {
      "type": "regular",
      "points": {
        "pA": {
          "x": 2465,
          "y": 2295
        },
        "pB": {
          "x": 2525,
          "y": 2295
        },
        "pC": {
          "x": 2525,
          "y": 2440
        },
        "pD": {
          "x": 2465,
          "y": 2440
        }
      }
    },
    {
      "type": "regular",
      "points": {
        "pA": {
          "x": 2575,
          "y": 2650
        },
        "pB": {
          "x": 2680,
          "y": 2650
        },
        "pC": {
          "x": 2680,
          "y": 2825
        },
        "pD": {
          "x": 2575,
          "y": 2825
        }
      }
    },
    {
      "type": "regular",
      "points": {
        "pA": {
          "x": 1510,
          "y": 2315
        },
        "pB": {
          "x": 1580,
          "y": 2280
        },
        "pC": {
          "x": 1620,
          "y": 2350
        },
        "pD": {
          "x": 1550,
          "y": 2385
        }
      }
    },
    {
      "type": "regular",
      "points": {
        "pA": {
          "x": 1535,
          "y": 2380
        },
        "pB": {
          "x": 1570,
          "y": 2360
        },
        "pC": {
          "x": 1605,
          "y": 2420
        },
        "pD": {
          "x": 1565,
          "y": 2440
        }
      }
    },
    {
      "type": "regular",
      "points": {
        "pA": {
          "x": 2135,
          "y": 3055
        },
        "pB": {
          "x": 2345,
          "y": 3055
        },
        "pC": {
          "x": 2345,
          "y": 3280
        },
        "pD": {
          "x": 2135,
          "y": 3280
        }
      }
    },
    {
      "type": "regular",
      "points": {
        "pA": {
          "x": 1870,
          "y": 2510
        },
        "pB": {
          "x": 1900,
          "y": 2510
        },
        "pC": {
          "x": 1900,
          "y": 2540
        },
        "pD": {
          "x": 1870,
          "y": 2540
        }
      }
    }
  ]
}
```

Lanchon, 2022-02-21:
it seems the sent data is ok. this could be related to the state the robot is in as per #1424.

> it would be interesting to know, how many entities are there in the map data sent by the API

i wouldn't know how to do that.

Hypfer, 2022-02-21:
Hm those are indeed 7 zones and 4 walls being sent.

Wow. Yeah
That's indeed a bug. How did this go unnoticed for so long?

Instead of counting coordinate pairs, we've counted every x and y seperately, which effectively halved the allowed virtual restrictions.

https://github.com/Hypfer/Valetudo/blob/411fab5496c9c2eba9f488d7d97a742073ddf3de/backend/lib/robots/roborock/capabilities/RoborockCombinedVirtualRestrictionsCapability.js#L45-L63

Whoops. Thanks for reporting

Lanchon, 2022-02-21:
cool! glad to be of help :)

commit cc5e37b6 (2022-02-21): fix(vendor.roborock): Fix roborock virtual restrictions counting vertices incorrectly

this fixes #1423
closed 2022-02-21 by Hypfer