Hypfer/Valetudo #1424: roborock s5 max: PersistentMapControlCapability: Invalid data, no reserve_map

Lanchon, 2022-02-21:
## Describe the bug
cannot enable persistent maps.

## To Reproduce
go to map management. persistent maps is disabled (although i the map is there and i think it is enabled). click to enable persistent maps. an error pops up:
> An error occurred while sending command to PersistentMapControlCapability: Invalid data, no reserve_map

### Vacuum Model
roborock s5 max latest fw (1566, 11/2021) just rooted

### Valetudo Version
2022.02.0 dde8f2ef28d8c7cd6b7456f6628595f9613dbf74

## Expected behavior
being able to enable persistent maps.

Hypfer, 2022-02-21:
Please do a factory reset and create a new map.
Previous usage with the official app probably left a state that Valetudo can't handle.

Lanchon, 2022-02-21:
thanks. i'll use the robot as it is. if you later change your mind and want to debug this, please reopen and i'll help. thanks!

Hypfer, 2022-02-21:
I can't debug this as I don't own an S5 max.
I also don't intend on buying one as full disassembly for rooting is not worth it IMO. Also money.

See also https://valetudo.cloud/pages/general/supported-robots.html#roborock_s5max "developer support" section.

Lanchon, 2022-02-21:
is there any way i can help debugging on my unit?

Hypfer, 2022-02-21:
`Invalid data, no reserve_map` does seem to be a response by the robots firmware itself.
It would be interesting to know what a reserve_map even is.

Apparently, you can either have lab_status 3 for three multi-maps(?) or lab_status 1 with a reserve_map that is some ID.

https://github.com/Hypfer/Valetudo/blob/411fab5496c9c2eba9f488d7d97a742073ddf3de/backend/lib/robots/roborock/capabilities/RoborockMultiMapPersistentMapControlCapability.js#L7-L10

As we're not sending that ID, the firmware is unhappy. That doesn't seem to be the case when initially enabling it though. It just happens when disabling multi-maps in the firmware.

Maybe the ID is the same that is fetched from the robots status response
https://github.com/Hypfer/Valetudo/blob/411fab5496c9c2eba9f488d7d97a742073ddf3de/backend/lib/robots/roborock/RoborockValetudoRobot.js#L313-L315

and then used in the map reset capability?
https://github.com/Hypfer/Valetudo/blob/411fab5496c9c2eba9f488d7d97a742073ddf3de/backend/lib/robots/roborock/capabilities/RoborockMultiMapMapResetCapability.js#L24-L30

It would be interesting to know what your robot currently responds to the `get_status` command.

You can see that by going to the logging view, setting the loglevel to trace and then just refreshing the main page.
You can then search for "map_status"

Lanchon, 2022-02-21:
there are a bunch of request/response pairs but they only differ in id and msg_seq

```
2022-02-21T13:13:17.292Z debug
>>> cloud: {"method":"get_status","params":{},"id":430}
2022-02-21T13:13:17.310Z debug
<<< cloud: {"id":430,"result":[{"msg_ver":2,"msg_seq":167,"state":8,"battery":100,"clean_time":71,"clean_area":1482500,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":3,"water_box_status":1,"fan_power":102,"dnd_enabled":0,"map_status":11,"is_locating":0,"lock_status":0,"water_box_mode":203,"distance_off":60,"water_box_carriage_status":1,"mop_forbidden_enable":1}],"exe_time":7}
```

Lanchon, 2022-02-21:
my robot at the time of rooting was configured for multiple persistent maps.

in case you don't know, this mode triggers a new map (max 4 i believe) when positioning is not successful. the previous map remains saved somewhere; local or in the cloud i don't know. the app lets you select a map as active. also the, robot switches automatically when positioning.

does valetudo support multiple maps on roborock? i did not see that option on my install.

Lanchon, 2022-02-21:
i tested:
```
~$ miiocli device --token 6c...a --ip $(dig roborocko.rod.lan +short) info
Model: roborock.vacuum.s5e
Hardware version: Linux
Firmware version: 3.5.8_1566
```

however this got me an empty answer:
```
$ miiocli device --token 6c...a --ip $(dig roborocko.rod.lan +short) raw_command get_properties '[{"did":39...3, "siid":18,"piid":19}]'
Running command raw_command
[]
```

and both commands took several seconds to complete (not the ip lookup).

however now they don't complete anymore, with:
```
ERROR:miio.miioprotocol:Got error when receiving: timed out
Error: No response from the device
```

what could be wrong? i know for sure i have mDNS (bonjour) disabled on my PC.

Lanchon, 2022-02-21:
rebooted and now it works.

```
get_status
Running command raw_command
[{'msg_ver': 2, 'msg_seq': 9, 'state': 8, 'battery': 100, 'clean_time': 71, 'clean_area': 1482500, 'error_code': 0, 'map_present': 1, 'in_cleaning': 0, 'in_returning': 0, 'in_fresh_state': 1, 'lab_status': 3, 'water_box_status': 1, 'fan_power': 102, 'dnd_enabled': 0, 'map_status': 11, 'is_locating': 0, 'lock_status': 0, 'water_box_mode': 203, 'distance_off': 60, 'water_box_carriage_status': 1, 'mop_forbidden_enable': 1}]

set_lab_status '[{"lab_status": 1, "reserve_map": 2}]'
Running command raw_command
['ok']

get_status
Running command raw_command
[{'msg_ver': 2, 'msg_seq': 10, 'state': 8, 'battery': 100, 'clean_time': 71, 'clean_area': 1482500, 'error_code': 0, 'map_present': 1, 'in_cleaning': 0, 'in_returning': 0, 'in_fresh_state': 1, 'lab_status': 1, 'water_box_status': 1, 'fan_power': 102, 'dnd_enabled': 0, 'map_status': 11, 'is_locating': 0, 'lock_status': 0, 'water_box_mode': 203, 'distance_off': 60, 'water_box_carriage_status': 1, 'mop_forbidden_enable': 1}]

get_lab_status
Running command raw_command
unknown_method

```

Hypfer, 2022-02-21:
Hey okay so the theory was correct. Thanks!

Lanchon, 2022-02-21:
the store persistent map setting in valetudo shows as enabled now (was showing as disabled). also FYI, my map was not lost.

what did {"lab_status": 1, "reserve_map": 2} do? select single persistent map mode + map #2 from the multimap set that existed before?

Hypfer, 2022-02-21:
> Apparently, you can either have lab_status 3 for three multi-maps(?) or lab_status 1 with a reserve_map that is some ID.

Turns out: This assumption is wrong. It's 1 and 3, because it's 00, 01 or 11. Roborock just loves using bit flags.

It's the same with the `map_status` field. That's a whole byte which contains of the mapId plus two flag bits

in this format:
`IIIIIISM`

I being all part of the map ID 0-63
S being a "segments present" flag
M being a "map present" flag

Hypfer, 2022-02-21:
This definitely improved the roborock codebase significantly. Thanks for reporting!

closed 2022-02-21 by Hypfer

Hypfer, 2022-02-21:
Apparently it's called save_map because it's the map you want to keep or phrased differently the map you want to save from being deleted when going to single-map-mode.

At least that's an explanation that makes some sense.

Lanchon, 2022-02-21:
thank you for this project! i wonder if you plan to support multimap. it is useful. and for that rare occasion where the robot gets lost and decides to trash your map, in multimap you can just undo that since it switches to a new slot instead of trashing it. that's really valuable.

right now i feel a bit unease when hacking the s5 max. it does not have a recovery method that i know of, so any misstep and i'm bricked. i think i should implement some recovery method.

Hypfer, 2022-02-22:
https://valetudo.cloud/pages/dev_faq.html#multiple-maps--multi-floor-

Lanchon, 2022-02-22:
ok i get it.

>  you can get a factory-new robot supported by Valetudo for less than 150€. If you own a multi-floor home...

i don't agree with that though. many parts of my home are not accessible by the robot. yes, another floor, but also the main balcony and 3 different patios. i wouldn't buy 7 robots (the other floor would need 2), nor would i leave robots exposed to the elements, nor would i install charging stations there.

maybe you can buy robots for 150 euro, but i can't because i don't live in a first world country. many countries -such as mine- are burdened by hefty import tariffs of outright bans. a friend had to hand-carry my robot from the USA in order for me to get it.

multimaps is a very useful feature for some, regardless of whether you choose to support it.

> Sometimes, the robot will report to the cloud that it won’t upload the requested map file again as instead the cloud should use file with ID XYZ.

if it can't be asked to re-upload then that particular bot would be left out of the feature, but no reason to not have the feature on other bots.

i'll probably go back to rooted stock because of the assurance that the feature will never be implemented. and it's fine, you don't like the feature and that's it. note that *i believe you* when you say it is tricky. (i just rooted and installed valetudo yesterday.) but, save for the new GUI for manual map selection, in my mind this looks like a simple case of detecting when the robot auto-switched maps and flush the valetudo caches. and maybe some hiccups here and there that users would have to accept in order to use multimaps.

obviously it is not as simple as i imagine, and i'd like to learn about the tricky details. do you happen to remember where the attempts or discussions regarding this took place?

btw, thanks again for valetudo, it looks and works amazingly well.

Hypfer, 2022-02-22:
Yes indeed that FAQ entry is written in an entirely german/western perspective.

Let me explain.
There are a lot of middle aged german men who absolutely have the money and the ability to buy multiple robots but refuse to do so, because they just feel like they don't want to.
That would be a valid choice, if those entitled fucks wouldn't also think that they're the ones who get to decide how I spend my free time. There has been so much nagging by these people constantly trying to force me to work on solving their bullshit issues that I just can't stand this topic anymore.

I agree that your case is solid, however at this time, I can't help you and I don't want to further discuss the pros, cons etc.

Hypfer, 2022-02-23:
After thinking about this some more, I've come to the very surprising conclusion that limiting your choices based on whether or not an option would benefit a group that you don't like is a just a petty thing to do. I'm sorry about that.
Furthermore, I'm sorry that the initial reaction you've received for your valid report mostly contained hostility based on experiences with groups of people that you're not even part of.

> but, save for the new GUI for manual map selection, in my mind this looks like a simple case of detecting when the robot auto-switched maps and flush the valetudo caches. and maybe some hiccups here and there that users would have to accept in order to use multimaps.

Yep. You're correct
In fact, the detect and clear caches thing already happens on roborock:

https://github.com/Hypfer/Valetudo/blob/b8b2f6b964f70be2cef95ddeb65b308bcac711a0/backend/lib/robots/roborock/RoborockValetudoRobot.js#L416-L422

Lanchon, 2022-02-23:
i'm glad you unlocked :)

you don't owe anything to anyone here. so just do what you want and say no to what you don't. some idiots will demand things from libre software developers. say no thanks, no explanation, move on. no point in getting emotionally involved... or trolled :)

and someone once said: never wrestle with a pig, you'll both get dirty and the pig likes it.

> the detect and clear caches thing already happens on roborock

thanks, i'll take a look as soon as i can.

commit e158ac4b (2022-02-21): fix(vendor.roborock): Properly parse and handle lab_status and map_status

this fixes #1424