Hypfer/Valetudo #1431: /mnt/data : No space left on device (on Roborock V1)

OhSoGood, 2022-02-26:
Hi,

I have an old version running on my Roborock V1 (Valetudo 0.4 if I'm not wrong). I've created and downloaded a new firmware image from Dustbuilder, and I cannot upload it to my vacuum. Any write on /mnt/data or any subfolders, even a simple file creation, leads to "No space left on device". 

Can anybody help me? Should I install the image the old way, with mirobo? Or can I use / to upload the image (I think / is volatile and and its content won't exist after a reboot, will it?).

Thank you!

My /etc/fstab:
```
# UNCONFIGURED FSTAB FOR BASE SYSTEM
tmpfs /tmp tmpfs size=30m 0 0
tmpfs /run/shm tmpfs size=100m 0 0
/dev/mmcblk0p1      /mnt/data/              ext4  defaults 0 0
/dev/mmcblk0p6      /mnt/default/               ext4  ro 0 0
/dev/mmcblk0p10     /mnt/updbuf/                ext4  defaults 0 0
/dev/mmcblk0p11     /mnt/reserve/               ext4  defaults 0 0

```

A df -h:
```
Filesystem       Size  Used Avail Use% Mounted on
/dev/root        494M  240M  229M  52% /
devtmpfs         202M  4.0K  202M   1% /dev
none             4.0K     0  4.0K   0% /sys/fs/cgroup
tmpfs             30M   24K   30M   1% /tmp
none              50M  220K   50M   1% /run
none             5.0M     0  5.0M   0% /run/lock
tmpfs            100M  2.3M   98M   3% /run/shm
none             100M     0  100M   0% /run/user
/dev/mmcblk0p11   16M  1.9M   13M  13% /mnt/reserve
/dev/mmcblk0p6    62M  5.3M   54M   9% /mnt/default
/dev/mmcblk0p1   1.6G  807M  662M  55% /mnt/data
/dev/mmcblk0p10  494M  238M  232M  51% /mnt/updbuf

```

broizter, 2022-02-26:
This is probably better suited for the discussion part since it's not a bug report. I would close this issue and posted your question there instead. https://github.com/Hypfer/Valetudo/discussions

OhSoGood, 2022-02-26:
Done. Thanks for the tip.

closed 2022-02-26 by OhSoGood