Hypfer/Valetudo #1435: Module recognision doesn't work

xathon, 2022-03-02:
## Describe the bug

The watertank and mop module don't seem to get recognized. Also, the dustbin removal is detected (the robot speaks), but it isn't shown in the Attachments section on the Home screen.
## To Reproduce

### Screenshots

[image]

### Vacuum Model

Roborock S50

### Valetudo Version

## Expected behavior

The attachments should get displayed properly.
## Additional context

Hypfer, 2022-03-02:
> Also, the dustbin removal is detected (the robot speaks), but it isn't shown in the Attachments section on the Home screen.

Unfortunately, there dustbin status is not reported by the robot. Release 2022.03 removed that confusing dustbin attachment.
As for the other two I'm confused since nothing about the code parsing the status changed since a whole year.

Hypfer, 2022-03-02:
I've managed to find the water tank of my S5 and can report that it does not detect any change to that at all, which is to be expected, given that it has no sensors for it.

I had forgotten just how basic the mop feature on that robot is. It's just a hollow piece of plastic that gets dragged along as well.

Hypfer, 2022-03-02:
Fixed with e4274d3d

closed 2022-03-02 by Hypfer

xathon, 2022-03-02:
> Unfortunately, there dustbin status is not reported by the robot. 

Ah, so the robot detects it internally but doesn't report it to the app?

> given that it has no sensors for it.

Ah. Well that makes sense then.

Thank you!

Hypfer, 2022-03-02:
> Ah, so the robot detects it internally but doesn't report it to the app?

Yep.

```
Sending cmd get_status with params[]
[
	{
		'msg_ver': 3,
		'msg_seq': 154,
		'state': 8,
		'battery': 77,
		'clean_time': 1,
		'clean_area': 0,
		'error_code': 0,
		'map_present': 1,
		'in_cleaning': 0,
		'in_returning': 0,
		'in_fresh_state': 1,
		'lab_status': 1,
		'water_box_status': 0,
		'fan_power': 102,
		'dnd_enabled': 0,
		'map_status': 3,
		'lock_status': 0
	}
]
```

it's just not part of the status nor is there another command that could pull the state