Hypfer/Valetudo #1472: Segment Management menu does not work

thor171, 2022-04-14:
## Describe the bug

In Segment Mangagement, it's not possible to edit segments because the UI says: "Editing segments requires the robot to be docked" even though the robot is docked. On the Home screen, the robot is docked.
## To Reproduce

After the robot made a persistent map, go to Segment Management and try to edit segments.
### Screenshots

[image]
[image]

### Vacuum Model

Xiaomi Mijia (Dreame) 1C
Firmware: 1110
Model: STYTJ01ZHM
### Valetudo Version

2022.03.1
## Expected behavior

Expected to be able to edit the segments.
## Additional context

Hypfer, 2022-04-14:
> On the Home screen, the robot is docked.

[image]

Are you sure about that?

thor171, 2022-04-14:
Hm, I just checked if it was, physically, docked. Maybe it was quite off. I will check that when I am at home.

thor171, 2022-04-14:
It did a cleanup again and went to the dock, but the state is idle. If it should say docked, then there might be another problem. I don't think it's a contact problem, because it's charging.

shepler007, 2022-04-18:
I just upgraded to 2022.3.1 and my bot is not generating segments.  It has been run twice and all I get is this big blue mapping.  My bot is a Roborock S5.
[image]

Hypfer, 2022-04-18:
@shepler007 https://github.com/Hypfer/Valetudo/discussions/1463

Different robot. Different issue

Hypfer, 2022-04-18:
Fixed with fdb51e7

closed 2022-04-18 by Hypfer