Hypfer/Valetudo #1475: Map changes don't get applied until cleaning is started (Dreame L10 Pro with Valetudo

bertvandepoel, 2022-04-21:
## Describe the bug

Because I have a freestanding, angled mirror in my bedroom, my map gets corrupted over time. Therefore, I do a new mapping run every few weeks to a month. Then I manually add back my exclusions and split up the rooms correctly. Weirdly enough, my cuts consistently don't show up until I start cleaning (but after the first cut, an error kind of confirms it has somehow happened in the background). Sometimes exclusions also have this issue. I've attached a video to show what happens.

## To Reproduce

1. Start a mapping pass
2. Accept map
3. try to cut map under map management

### Screenshots

https://youtu.be/h1k69EtMRwE

### Vacuum Model
Dreame L10 Pro

### Valetudo Version

2022.03.1

## Expected behavior

I would expect that changes would show up after a few seconds or at least wouldn't require me to start cleaning

## Additional context

Technically if I could upload a saved map, I wouldn't run into this problem, but I expect that's not in the cards for now.

Hypfer, 2022-04-21:
Not a bug in Valetudo. We just poll the current map all the time
Why the robot sometimes decides to not upload a recent map is unclear and nothing we can do anything about.

closed 2022-04-21 by Hypfer

stugr, 2022-04-22:
I'm having the same issue with my Dreame L10 pro. Have to make a single map change, start the robot, send it home, refresh the page, rinse and repeat

bertvandepoel, 2022-04-22:
It's really weird, a friend of mine with the same HW revision and firmware version doesn't have this problem. It's good to know that you do stugr. No idea what we can do about it though :/