Hypfer/Valetudo #1482: Dustcloud connection fails when WiFi's DHCP does not assign default route

TheAssassin, 2022-04-29:
*This issue had been fixed before: https://github.com/Hypfer/Valetudo/pull/508*

## Describe the bug

While making the switch to original Valetudo on my Xiaomi gen 1 robot (using a dustbuilder generated image), I made it join my IoT WiFi (worked fine), but Valetudo could not control the robot (the UI didn't show any error messages, just wasn't responsive, the majority of the features could not be controlled but showed some loading state). I couldn't find a log file, so I ran the `valetudo` binary manually, I found this:

```
[...]
[2022-04-29T20:59:22.699Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2022-04-29T20:59:22.713Z] [INFO] Webserver running on port 80
[2022-04-29T20:59:22.721Z] [INFO] FDSMockServer running on port 8079
[2022-04-29T20:59:22.750Z] [INFO] Bonjour service "Valetudo V1 Web" with type "http" started
[2022-04-29T20:59:22.752Z] [INFO] Bonjour service "Valetudo V1" with type "valetudo" started
[2022-04-29T21:00:22.452Z] [WARN] Unable to reach vacuum. Giving up after 100 tries
[2022-04-29T21:01:17.121Z] [WARN] Unable to reach vacuum. Giving up after 100 tries
[2022-04-29T21:01:17.124Z] [WARN] Error while fetching statistics RetryWrapperSurrenderError [MiioTimeoutError]: request timed out:{"method":"get_status","params":{},"id":102102}
    at RetryWrapper.retryHandshakeHelper (/snapshot/Valetudo/backend/lib/miio/RetryWrapper.js:181:19)
    at doRetry (/snapshot/Valetudo/backend/lib/miio/RetryWrapper.js:130:22)
    at /snapshot/Valetudo/backend/lib/miio/RetryWrapper.js:148:25
[...]
```

I immediately remembered the fix I contributed to the non-fork, which @pidator integrated into this repository as well. Checking `iptables-save`, I found that the spoofing is still done by redirecting that Xiaomi IP to localhost:

```
[...]
# Generated by iptables-save v1.4.21 on Sat Apr 30 05:26:42 2022
*nat
:PREROUTING ACCEPT [279:17080]
:INPUT ACCEPT [270:16224]
:OUTPUT ACCEPT [83:5394]
:POSTROUTING ACCEPT [106:6774]
-A OUTPUT -d 203.0.113.1/32 -p tcp -m tcp --dport 80 -j DNAT --to-destination 127.0.0.1:8053
-A OUTPUT -p udp -m udp --dport 8053 -j DNAT --to-destination 127.0.0.1:8053
[...]
```

This rule still only works when the IP is in some range the kernel has a route for. In my special network, there is no default route, since there is no access to the Internet. The DHCP server therefore also does not assign a default route.

The fix is simple: restore #508.

I'd like to contribute a fix yet again, since I really like Valetudo, and want it to work out of the box. I don't see any use of `rc.local` any more, though. Where do you think adding the route could be integrated? Since `valetudo` is running as `root`, it could perform the call on startup for instance.

## To Reproduce

Use Valetudo to join a WiFi network whose DHCP server does not assign a default route.

Alternatively, `ip route del default dev wlan0` should provide a similar experience.

### Screenshots

### Vacuum Model
Xiaomi gen 1 (pre 2020)

### Valetudo Version
2022.03.01

rimrul, 2022-04-30:
It was removed in 5492e4d and it seems like [Hypfer would prefer this to go into dustbuilder](https://github.com/Hypfer/Valetudo/commit/5492e4dae0923577a5546a09a8b9e56918f88d12#commitcomment-47194693)

TheAssassin, 2022-04-30:
I guess it make sense for this iptables rule to be added wherever the other rules are. Indeed it's more of a deployment issue. I'll probably send them a PR then. But let's hear @Hypfer's opinion, too.

Hypfer, 2022-04-30:
Definitely out of scope for Valetudo

closed 2022-04-30 by Hypfer

TheAssassin, 2022-05-10:
*(Please don't chase me off the yard for adding a comment to this already-closed issue.)*

For the record (for anyone else looking for a solution): [vacuum firmware builder](https://github.com/zvldz/vacuum/blob/2bd29707e01671de35cf1042afd5abfaaff284d2/custom-script/files/valetudo/etc/rc.local#L16) has the solution implemented, this just affects [dustbuilder](https://dustbuilder.dontvacuum.me/). I couldn't find their source code easily, seems like they may use an outdated fork of that vacuum project.