Hypfer/Valetudo #1521: Wifi disconnects shortly after clean commences

lukyjay, 2022-07-19:
## Describe the bug

Roborock works fine when connected to the dock. Once I start cleaning, the wifi disconnects and the blue wifi light blinks rapidly. It will reconnect after a few hours.

I have full coverage of Wifi and this issue has only presented itself in the last few updates (sorry, not sure which one)

Is there a way I can access logs to troubleshoot this? I can connect via SSH after it reconnects and see if there's anything useful
## To Reproduce

The Wifi always disconnects after a few minutes of leaving the dock.

### Vacuum Model

Roborock S5

### Valetudo Version

2022.06.00
## Expected behavior

Wifi to stay connected during clean.
## Additional context

I lost my SSH key and had to factory reset the robot around a month ago. This issue was not resolved through factory reset.

It wasn't happening last year. It's only started happening in the last few updates.

Hypfer, 2022-07-19:
Hi there,

your issue has been closed, because it is either a support request or something else that would usually be handled by 1st level support.

Since this is not a commercial project, but rather just something I've built for myself and then shared with the world for free, I really can't provide said support.

Therefore, please direct this request somewhere else, where other users might be able to help you.
The Telegram group would be one option but certainly isn't the only one.

Thank you for understanding as well as your interest in the project.

lukyjay, 2022-07-19:
Do you know where the relevant log locations are? Maybe they will help finding the cause

lukyjay, 2022-07-20:
Ok sorry I've mislead you but it appears wifi is not dropping it's just the web interface from what I can tell.

I am connected via SSH but the web interface (Valetudo) is not working

Logs from /tmp/valetudo.log:
```
[2022-07-20T03:22:48.313Z] [INFO] Network state changed
[2022-07-20T03:22:48.322Z] [INFO] Restarting NetworkAdvertisementManager
[2022-07-20T03:22:48.376Z] [INFO] Valetudo can be reached via: valetudo-nutritiouswaryboar.local
[2022-07-20T03:22:48.417Z] [INFO] Bonjour service "Valetudo NutritiousWaryBoar Web" with type "http" started
[2022-07-20T03:22:48.418Z] [INFO] Bonjour service "Valetudo NutritiousWaryBoar" with type "valetudo" started
[2022-07-20T03:23:18.402Z] [INFO] Network state changed
[2022-07-20T03:23:18.404Z] [INFO] Restarting NetworkAdvertisementManager
[2022-07-20T03:23:18.425Z] [INFO] Valetudo can be reached via: valetudo-nutritiouswaryboar.local
[2022-07-20T03:23:18.466Z] [INFO] Bonjour service "Valetudo NutritiousWaryBoar Web" with type "http" started
[2022-07-20T03:23:18.467Z] [INFO] Bonjour service "Valetudo NutritiousWaryBoar" with type "valetudo" started
[2022-07-20T03:24:18.459Z] [INFO] Network state changed
[2022-07-20T03:24:18.461Z] [INFO] Restarting NetworkAdvertisementManager
[2022-07-20T03:24:18.484Z] [INFO] Valetudo can be reached via: valetudo-nutritiouswaryboar.local
[2022-07-20T03:24:18.510Z] [INFO] Bonjour service "Valetudo NutritiousWaryBoar Web" with type "http" started
[2022-07-20T03:24:18.511Z] [INFO] Bonjour service "Valetudo NutritiousWaryBoar" with type "valetudo" started
[2022-07-20T03:24:20.513Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2022-07-20T03:24:20.517Z] [ERROR] RobotMQTTHandle: Error while polling the robots state MiioTimeoutError: request timed out:{"method":"get_status","params":{},"id":1307}
    at Object.onTimeoutCallback (/snapshot/Valetudo/backend/lib/miio/MiioSocket.js:193:32)
    at Timeout._onTimeout (/snapshot/Valetudo/backend/lib/miio/MiioSocket.js:199:53)
    at listOnTimeout (node:internal/timers:564:17)
    at process.processTimers (node:internal/timers:507:7)
[2022-07-20T03:24:48.504Z] [INFO] Network state changed
[2022-07-20T03:24:48.506Z] [INFO] Restarting NetworkAdvertisementManager
[2022-07-20T03:24:48.524Z] [INFO] Valetudo can be reached via: valetudo-nutritiouswaryboar.local
[2022-07-20T03:24:48.557Z] [INFO] Bonjour service "Valetudo NutritiousWaryBoar Web" with type "http" started
[2022-07-20T03:24:48.558Z] [INFO] Bonjour service "Valetudo NutritiousWaryBoar" with type "valetudo" started
[2022-07-20T03:24:58.533Z] [INFO] Dummycloud connected
```

Before reboot, whilst valetudo isn't working:
```
[root@rockrobo rockrobo]# ps
  PID USER       VSZ STAT COMMAND
    1 root      2264 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    6 root         0 SW   [migration/0]
    7 root         0 SW   [watchdog/0]
    8 root         0 SW   [migration/1]
   10 root         0 SW   [ksoftirqd/1]
   11 root         0 SW   [watchdog/1]
   12 root         0 SW   [migration/2]
   14 root         0 SW   [ksoftirqd/2]
   15 root         0 SW   [watchdog/2]
   16 root         0 SW   [migration/3]
   18 root         0 SW   [ksoftirqd/3]
   19 root         0 SW   [watchdog/3]
   20 root         0 SW<  [cpuset]
   21 root         0 SW<  [khelper]
   22 root         0 SW   [kdevtmpfs]
   23 root         0 SW<  [netns]
   28 root         0 SW   [sync_supers]
   29 root         0 SW   [bdi-default]
   30 root         0 SW<  [kblockd]
   31 root         0 SW   [sytem]
   32 root         0 SW   [khubd]
   34 root         0 SW<  [cfg80211]
   35 root         0 DW   [usb-hardware-sc]
   36 root         0 SW   [cfinteractive]
   37 root         0 SW   [kdsm]
   38 root         0 SW<  [rpciod]
   39 root         0 SW   [kswapd0]
   40 root         0 SW   [fsnotify_mark]
   41 root         0 SW<  [nfsiod]
[... 135 lines cut ...]
  368 root     29360 S    /usr/bin/adbd
  369 root      1868 S    /usr/sbin/dropbear -F
  370 root      2268 S    /sbin/getty -L ttyS0 115200 vt100
  427 root     17484 S <  rrlogd /opt/rockrobo/rrlog/rrlog.conf
  454 root      119m S <  rr_loader -d
  524 root     19480 S <  RoboController
  526 root     27680 S <  AppProxy
  528 root     10304 S <  wlanmgr
  533 root      3728 S <  miio_client -d /mnt/data/miio -l 0
  539 root     13992 S <  rriot_tuya -d
  693 root         0 SW   [kworker/0:2]
  727 root         0 RW   [ksdioirqd/mmc1]
  770 root         0 RW   [RTW_XMIT_THREAD]
  771 root         0 SW   [RTW_CMD_THREAD]
  772 root         0 SW   [RTWHALXT]
  773 root         0 SW   [RTWHALXT]
  775 root      2864 S <  wpa_supplicant -D nl80211 -B -i wlan0 -C /var/run/wp
  780 root      2288 S <  {miio_client_hel} /bin/bash /opt/rockrobo/miio/miio_
  980 root         0 SW   [kworker/3:2]
 1005 root      1296 S <  miio_recv_line
 1117 root      2264 S <  /sbin/udhcpc -S -b -i wlan0
 1282 root      1936 R    /usr/sbin/dropbear -F
 1285 root      2268 S    -ash
 1292 root      2268 R    ps
```

lukyjay, 2022-07-27:
First factory reset didn't solve it but the second one did. All fixed now.

Shouldn't the watchdog have restarted the valetudo binary, or is that unrelated? 

Feel free to close the issue if you think this is a once off. Apologies for any confusion, have a newborn at home 😴😴

lukyjay, 2022-07-27:
I spoke too soon, it just happened again. I was recreating the map and it took longer to crash. Could it be due to a large map size or something?

Hypfer, 2022-07-27:
Are there any additional logs?

lukyjay, 2022-07-27:
I'm going to try a fourth firmware reset, this time I won't restore my conf file.

Is it just the /tmp/valetudo.log file I should pay attention to?

lukyjay, 2022-07-27:
Fresh reset with no touching of SSH or backups. On an SSID which is 2.4ghz only. Another interface loss but SSH working fine. This time Valetudo is still running:
[root@rockrobo ~]# ps | grep valetudo
 9664 root      2268 S    grep valetudo

Log shows a few errors due to connectivity this time:
https://pastebin.com/SiRs8EKe

To try the nightly, do I just download the armv7 binary and replace /usr/local/bin/valetudo ?

Hypfer, 2022-07-27:
yes

lukyjay, 2022-08-02:
Just confirming it still happens under the nightly you shared. 

Has there been no other reports of this? It's really strange that it would only affect one user. It implies user (me) error. 

To confirm I have this right, I factory reset by hitting the physical button on the robot. I then join her wifi network and launch miioota software to flash the image I downloaded from dustbuilder, then wait for her to reboot and finalise and rejoin my normal wifi network.

Please let me know if there's a way to enable debug logging.

Hypfer, 2022-08-03:
> Has there been no other reports of this?

Not at all.

That reset button does not factory reset on roborock though. It just resets the wifi credentials
Factory resetting the S5 can be done like this:
https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Factory-Reset#gen2

lukyjay, 2022-08-03:
> That reset button does not factory reset on roborock though. It just resets the wifi credentials Factory resetting the S5 can be done like this: https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Factory-Reset#gen2

Sorry that's what I meant by reset button 😃

Dmesg shows a bunch of wifi disconnects. I've disabled all of the roaming on my Omada APs and will see if that fixes it. 

It shouldn't, since the network reconnects and SSH works. Maybe the valetudo binary crashes when the wifi disconnects? Will let you know if that fixes it.

Hypfer, 2022-08-03:
Since there's nothing in any log, I somewhat doubt that it crashes. Are you 110% sure that it stops running and not just becomes unavailable due to networking issues?

lukyjay, 2022-08-03:
Unless I'm missing it, it isn't showing in the first 'ps' output I shared above once the web interface stopped working. It's possible that isn't the cause, but other network connectivity continues to work (SSH fr example).

sky321, 2022-08-04:
Sorry that I‘m late to the party but had the same several times before. I‘m not into deep here but can confirm that the UI stop working but Robot react on webhooks via REST. In my case it‘s a L10. I flashed the latest 2022.8.0, lets see if it continues.

Hypfer, 2022-08-04:
Without any logs whatsoever, there is nothing that can be done here.
I'd expect to see either the kernel logging a segfault of the binary (very unlikely!) or (much more likely) the valetudo log logging at least _something_.

This issue played out exactly as expected and that's also exactly why my first reaction was to just close it to be done with it.
Now we're just guessing and throwing together random things that likely aren't even related to each other.

Anyway, two days ago, someone reported that valetudo crashed (with logs!) when using an old version of the palemoon firefox fork. It could be that the fix for that might also fix your issue so please again try the latest nightly.

closed 2022-08-04 by lukyjay