Hypfer/Valetudo #1522: No MQTT ManualControlCapability for RoborockV1, 2022.06.0

thle, 2022-07-19:
## Describe the bug
I see the manual control in Web UI, but when I look at MQTT Explorer there is no ManualControlCapability, just:
`MapData,FanSpeedControlCapability,BasicControlCapability,ConsumableMonitoringCapability,ZoneCleaningCapability,GoToLocationCapability,LocateCapability,CurrentStatisticsCapability,WifiConfigurationCapability,StatusStateAttribute,BatteryStateAttribute`

## To Reproduce
1. Installed 2022.06.0
2. Connected to openHAB 3.1.1 MQTT Binding and Broker
3. launch MQTT Explorer

## Screenshot
[image]

### Vacuum Model
RoborockV1

### Valetudo Version
2022.06.0

## Expected behavior
ManualControlCapability available

Hypfer, 2022-07-19:
Correct. Not every capability is exposed to MQTT.

In case of the ManualControlCapability that's because it needs a direct line of sight to be operated, meaning that the MQTT interface, which is used primarily for automations, doesn't make sense for this type of functionality.

closed 2022-07-19 by Hypfer

thle, 2022-07-19:
That's odd because I used this kind of automation to move the robot from under a furniture for maintenance. Is there another solution?