Hypfer/Valetudo #1527: Manual Control improvements

Hypfer, 2022-07-28:
Currently, Valetudo only features very basic support for manual control.

There's the `ManualControlCapability`, which allows for only up to four commands:
```
- FORWARD
- BACKWARD
- ROTATE_CLOCKWISE
- ROTATE_COUNTERCLOCKWISE
```

Most supported robots however support much higher resolution manual movement commands.
Both Dreame and Roborock accept control messages featuring two floats for velocity and angle. 

The Idea here would be to introduce a new `HighResolutionManualControlCapability`, which provides an interface that is much more sophisticated than the `ManualControlCapability`. This new capability might co-exist with the old one. That is TBD

Apart from that, the UI could also use some work:

- General usability and visual improvements
- Support for WASD/Arrow Keys
- Support for Gamepad input

would be a few ideas

hadess, 2024-08-02:
> Anyone else see our robots racing against each other? 😁

For that, you would also need a live feed from camera to do FPV races ;)

hadess, 2024-08-02:
One functionality that would be useful to me (as opposed to just funny to have like the above live view) would be to be able to move the robot without starting the vacuuming function.

My robot lives under a kitchen island ([IKEA VARDE](https://github.com/user-attachments/assets/b0b98ca0-6cd7-4276-84a7-72c6953d035a)) and I need to either crawl under it or use the manual controls to get access to the dust bin.

Ideally, I could make the robot move 50cm forward, lift the access door, and empty the bin without it starting the vacuuming, or yelling at me that it needs its bin reseated. Maybe a vacuuming toggle switch?

The last thing would be for the home button to also appear on the manual control page, so I could do all this from a single page.

(Let me know if you want me to move this to a different issue)

ChuckyCheegles, 2024-10-22:
Can we also get MQTT support for manual control? I want to add buttons to Home Assistant that will manually move the vacuum if it gets stuck or something but the only way I've found to do it is a sort of hacky way to interface with the REST API and there's very little documentation(I understand there's a lack of docs because end users aren't supposed to interface with it and it might break the frontend).

closed 2024-11-01 by Hypfer