Hypfer/Valetudo #1530: Convert ValetudoMaps to 3d/cad/building information formats

Hypfer, 2022-07-28:
It might be useful to have some kind of standalone tool that takes a ValetudoMap json and converts it to standard formats that one would want to use when modelling a building.

I honestly have absolutely zero knowledge what exactly the data formats and/or requirements would be, but I heard of people that modeled their new homes in google sketchup to arrange furniture in advance.
Being able to let a standard 200€ robot vacuum _that you already have_ do all the heavy lifting for you would be pretty neat.

Also, who knows what else one might be able to do with that data in the right format.
Create some floorplans for other purposes maybe?

Carko, 2022-07-31:
The idea sounds nice at the first glance, but I think there would be just too much noise for the results to be useful.

I have just looked at a few hundred vacuum maps on google and there were _maybe_ a few of them that you could say could be converted to CAD and be useful for anything at all, even assuming you do filtering of the artifacts and small objects.

The more furniture you have on the floor, more of the lines would need to be removed afterwards and redrawn manually. E.g. in a kitchen, it's quite common that there's no wall visible at all from the floor level. 

Also if the dimensions are slightly off (they usually are) it's difficult to correct them as the whole rooms are shifted, so you can not just linearly scale the whole map

But the main point is that all common CAD application support using images as backgrounds to draw upon, so you can easily place a screenshot from Valetudo, scale it and draw the floor plan on top of it in a fraction of the time it would take you to remove all the garbage from the vacuum map.

ingoschi, 2022-08-07:
I think the idea is to use the robot to get a map of an empty apartment before furniture is added.

sca075, 2023-08-03:
I developed one integration for home assistant that grab the maps from the robots with Valetudo firmware. To extract the map from the MQTT image, isn't a simple a thing but in my project works. I currently support two vacuums, but I can implement also different models to this camera application. If the task is to get the floor level of the map I can implement a function that produce the image of the floor plan. If you are interested you can give it a try.

closed 2023-08-20 by Hypfer