Hypfer/Valetudo #1548: Updater offers older Version

kanonet, 2022-08-10:
## Describe the bug

I upgraded my Dreame L10 Pro to Valetudo 2022.08.0 and FW 1138. "System Information" is correctly displaying those newly installed versions. But the integrated "Updater" offers me to "upgrade" to 2022.06.0 again. This version is obviously older then the installed one. Since FW 1138 requires Valetudo 2022.08.0 to run, downgrading would actually be harmful.

## To Reproduce

1. Get a Dreame L10 Pro with Valetudo 2022.08.0 and FW 1138
2. Check for Update

## Expected behavior

No update should be offered when the robot already is on latest version. Especially no outdated version.

Hypfer, 2022-08-10:
[image]

Can't reproduce. Are you 100% sure that you're running `2022.08.0`? 
Note that the updater always just updates to the next chronological release so if you're seeing `2022.06.0` now, you're likely still on `2022.05.1`

kanonet, 2022-08-10:
I just got an idea: when creating the new FW image on DustBuilder I checked "Prepackage valetudo 2022.08.0". Do you think this caused a conflict with the one that was already installed and upgraded? If this is the cause, we should state this in the documentation somewhere.

Hypfer, 2022-08-10:
> Do you think this caused a conflict with the one that was already installed and upgraded?

Nope. That install script just moves the binary included in the tar file to `/data/valetudo`.

Did you try a hard refresh with ctrl + f5? Did you try rebooting the robot?
Do you have any caching in your local network?

All the updater does is fetching this json https://api.github.com/repos/Hypfer/Valetudo/releases and sorting it by `published_at`.
It then takes this list and searches for the index of the element where `tag_name` matches the one from the baked-in package.json.
If it finds it, it takes index-1. Else it just takes the first element

I don't see anything in there that could cause behavior as described here unless the API response from github looks different.
If for some reason 2022.06.0 had a newer published_at date, then it would indeed be sorted before 2022.08.0.

Checking here though, that doesn't seem to be the case.
In any case, that would not be a bug in Valetudo but in the github api.

kanonet, 2022-08-10:
A hard refresh did not change anything and I dont think anything in my network is responsible. Im using valetudo for 9 months now and did never see this Problem before. While installing the new FW a few days ago, the robot did a reboot. Since then it automatically did a quiet/soft restart every 24h.
Now I just forced a hard restart by pressing the hardware button - and **the Problem is solved** now - "_You are already running the latest version of Valetudo_".
Since it is solved for now, this issue can get closed. I still wonder what caused this and why the earlier restarts (see above) did not suffice. If it somehow happens again, I will report back. Thanks for your help!

closed 2022-08-10 by Hypfer