Hypfer/Valetudo #1549: Valetudo Web interface is not reachable after WiFi disconnect and reconnect

picakia, 2022-08-14:
## Describe the bug

Valetudo interface is not reachable on local IP address after robot has been disconnected from WiFi access point for about 4-5 minutes. SSH access works and restarting Valetudo service via `/etc/init/S11valetudo restart` fixes the problem. 

## To Reproduce

### Screenshots

[image]
[image]
[image]

### Vacuum Model

Roborock S5

### Valetudo Version

2022.8.0

## Expected behavior

After robot has been reconnected to WiFi access point after being disconnected for longer than 5 minutes Valetudo interface should work normally. 

## Additional context

I used phone access point for testing this behavior. 
[Valetudo log file](https://github.com/Hypfer/Valetudo/files/9333260/valetudo.log)

Hypfer, 2022-08-14:
Strange. Sounds similar to #1521
Unfortunately, it is impossibly to understand the issue from the information provided as they're basically just "hey, broken!".

What confuses me the most is that there is nothing in the Valetudo log.
Do you see something in the `dmesg` output or any other logfile you might find in the FS?

Hypfer, 2022-08-14:
Trying out the steps to reproduce with my S5 didn't work out.
I've tried it two times and Valetudo is still running fine

fabir-git, 2022-08-15:
> Strange. Sounds similar to #1521 Unfortunately, it is impossibly to understand the issue from the information provided as they're basically just "hey, broken!".
>
> What confuses me the most is that [...]

I am having similar issues (loosing wifi some times after cleaning and some times random).
Here ist the dmesg output after the last time the issue occured:

`
[root@p2029_release:/data/log]# dmesg
[    0.989113] deviceless supply vcc-nand not found, using dummy regulator
[    0.989354] nand:get voltage vcc-nand ok:ffffffc01cef5200
[    0.989361] nand0_regulator2 none
[    0.989384] deviceless supply none not found, using dummy regulator
[    0.989549] nand:get voltage vcc-io ok:ffffffc01cef5980
[    0.989554] nand:has already get voltage
[    0.989562] nand_clk_request
[    0.989611] nand_clk_request: get pll rate 1200000000HZ
[    0.989779] [ND]Reset NDFC start 0  0
[    0.989786] [ND]Reset NDFC end 0     0
[    0.989852] [NI]find the device:FS33ND04GS108TF10
[    0.989861] [NI]rawnand support chip 0: ec dc 10 95 56 c6 ec dc
[    0.989908] [NE]rawnand not support chip 1: 03 03 03 03 03 03 03 03
[    0.989923] id_number_ctl is no used
[    0.989929] [ND]build nand physical connection ok...
[    0.989934] [ND]CE:
[    0.989939] [ND]0 [ND]
[    0.989943] RB:
[    0.989948] [ND]0 <7>[    0.989953] [ND]
[    0.990181] [ND]check scan data, first_check 4 1!
[    0.990191] [ND]329 interface change ddr_type:0 sclk0:40 sclk1:80 sdr_edo_bak                                                                         :1 ddr_edo_bak:1 ddr_delay_bak:0 !
[    0.990199] [ND]mode 6: unchanged, rawnand set to sdr
[    0.990204] [ND]mode 6
[    0.990217] [ND]_get right timing para, set edo to 1 for sdr nand.
[    0.990224]  generic special init
[    0.990229] [ND]nand_physic_info_read start!!
[    0.990929] [ND]physic_info_get_one_copy start!!
[    0.991925] [ND]block 16 page 40
[    0.992046] [ND]block 16 page 41
[    0.992167] [ND]block 16 page 42
[    0.992287] [ND]block 16 page 43
[    0.992408] [ND]block 16 page 44
[    0.992528] [ND]block 16 page 45
[    0.992648] [ND]block 16 page 46
[    0.992769] [ND]block 16 page 47
[    0.992889] [ND]block 16 page 48
[    0.993010] [ND]block 16 page 49
[    0.993130] [ND]block 16 page 50
[    0.993251] [ND]block 16 page 51
[    0.993371] [ND]block 16 page 52
[    0.993491] [ND]block 16 page 53
[    0.993612] [ND]block 16 page 54
[    0.993732] [ND]block 16 page 55
[    0.993914] [ND]physic info copy is ok
[    0.993934] [ND]rawnand_sp_chips_init 2459
[    0.993941] [NI]==============show_nctri======================
[    0.993946] [NI]channel_id:            0
[    0.993951] [NI]type:                  1
[    0.993956] [NI]chip_cnt:              1
[    0.993961] [NI]chip_connect_info:     1
[    0.993966] [NI]rb_connect_info:       1
[    0.993970] [NI]ce:
[    0.993975] [NI]0 [NI]
[    0.993979] rb:
[    0.993984] [NI]0 <6>[    0.993988] [NI]
[    0.993993] [NI]dma_type:              1
[    0.993998] [NI]dma_addr:              5cf0f800
[    0.994004] [NI]write_wait_rb_before:  1
[    0.994008] [NI]write_wait_rb_mode:    1
[    0.994013] [NI]rb_ready_flag:         0
[    0.994018] [NI]dma_ready_flag:        1
[... 411 lines cut ...]
[    4.366414] gpio-0 (?): gpiod_request: status -16
[    4.366419] gpio-0 (?): gpiod_request: status -16
[    4.366425] gpio-0 (?): gpiod_request: status -16
[    4.366430] gpio-0 (?): gpiod_request: status -16
[    4.366435] gpio-0 (?): gpiod_request: status -16
[    4.613339] of_get_named_gpiod_flags: parsed 'sensor1_pwm1' property of node                                                                          '/soc@03000000/vind@0/sensor@1[0]' - status (0)
[    4.613354] of_get_named_gpiod_flags: parsed 'sensor1_pwm2' property of node                                                                          '/soc@03000000/vind@0/sensor@1[0]' - status (0)
[    4.613366] of_get_named_gpiod_flags: parsed 'sensor1_pwm3' property of node                                                                          '/soc@03000000/vind@0/sensor@1[0]' - status (0)
[    4.613378] of_get_named_gpiod_flags: parsed 'sensor1_vsync' property of node                                                                          '/soc@03000000/vind@0/sensor@1[0]' - status (0)
[    4.614132] [sc031gs_mipi]PWR_ON!
[    4.635634] [sc031gs_mipi]PWR_OFF!
[    5.273488] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    5.555065]
[    5.555065] insmod_device_driver
[    5.555065]
[   10.404119] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   15.384592] [ND]gc_one ok! from: 1 [80 36 120] @nand 756
[   38.777288] unexport_store: status -22
[   88.118546] [sc031gs_mipi]PWR_ON!
[   88.146256] [sc031gs_mipi]sensor_s_stream on = 1, 640*480 fps: 30 code: 3007
[   88.185588] [sc031gs_mipi]s_fmt set width = 640, height = 480
[  157.842230] [ND]gc_one ok! from: 1 [78 36 120] @nand 756
[  162.308682] [sc031gs_mipi]sensor_s_stream on = 0, 640*480 fps: 30 code: 3007
[  162.312031] [sc031gs_mipi]PWR_OFF!
`

picakia, 2022-08-16:
I reproduced the issue again, right now I'm not sure what causes this behavior since it happens only sometimes. It looks like when this happens there are a lot of get_status timed out in valetudo log. I'm attaching dmesg log as well. 

From my testing it happened when I:
1. Connected to AP,
2. Removed my map
3. Started new mapping
4. Finished mapping
5. Set up segments
6. Turned robot off and left it off overnight
7. Turned it on by placing it on the dock
8. Turning phone AP on after robot finished booting
9. Trying to connect to valetudo which resulted in valetudo interface showing loading symbols on every switch and menu 
[image]
10. A little bit later Valetudo interface is not reachable in any way

In addition if you want and have some free time we can organize debugging session via Telegram where I can give you ssh access to the robot while it is in bugged state since ssh always works. 
Thanks for quick response and for making awesome software for our vaccums! 
[valetudo.log](https://github.com/Hypfer/Valetudo/files/9349927/valetudo.log)
[dmesg.log](https://github.com/Hypfer/Valetudo/files/9349928/dmesg.log)

Hypfer, 2022-08-17:
Thank you for looking into this furhter.

Unfortunately, it is unlikely that this issue can be solved in a reasonable amount of time.
The logs attached don't show anything too concerning. No segfault of a valetudo binary nor it shutting down.
I have absolutely no idea what is going on there.

Regarding the `get_status` timeouts, I'm also at a loss. It could be the miio_client not running because there is no wifi connection but it also could be the miio_client rejecting our messages because it thinks that they are duplicates even when they aren't. I've seen that behavior before but I wasn't ever able to understand why it is happening.

As the miio_client is distributed as a binary, we can only resort to ghidra to see what it does.
Building our own miio_client could be a solution to that but no one yet bothered to do that.

Apart from that, there's also an issue with the retry logic in Valetudo where it could handle the miio_client not responding more gracefully. No idea when and how though.

As your issues seem to be caused by an unstable wifi connection, I'd suggest just fixing that as it is a few orders of magnitude easier than figuring out what strange edge case it is triggering. Furthermore, not having a stable wifi connection isn't a desirable state to permanently have so it's not like it would be some workaround.

closed 2022-08-17 by Hypfer