Hypfer/Valetudo #1555: Dreame D9 crashing when starting a clean

mals, 2022-08-21:
The vacuum appears to be crashing within a few seconds of starting a clean. What I am observing is that the vacuum stops working and the robot stops moving. I am no longer able to ping the vacuum or ssh to it. This happens within the first 30 seconds of starting a clean but usually within the first 5 seconds.

## To Reproduce
Press the power button to start a clean (or start a clean through the Valetudo UI)

### Screenshots

### Vacuum Model
Dreame D9

### Valetudo Version
2022.08.0
Firmware 1132

## Expected behavior
The vacuum should not crash and continue cleaning

## Additional context
This is an initial installation of Valetudo on a newly rooted Dreame D9. I am attempting to downgrade to FW 1100 with Valetudo 2022.08.0.

mals, 2022-08-21:
Downgrading to FW 1100 appears to have fixed the problem.

Hypfer, 2022-08-21:
This issue is caused by the MCU Firmware being too old.

To fix this, try the following:
- Make sure that the robot is docked
- Download the dustbuilder firmware update package onto the robot
- run install-manual.sh
- Wait for it to reboot
- Download the dustbuilder firmware update package onto the robot
- run install.sh
- Wait for it to reboot
- Download the dustbuilder firmware update package onto the robot
- run install-mcufw.sh
- Wait for it to reboot

If it is not yet fixed, try repeating the install-mcufw.sh step a few times