Hypfer/Valetudo #1563: MQTT set topics not working

systemofapwne, 2022-08-24:
### Discussed in https://github.com/Hypfer/Valetudo/discussions/865

<div type='discussions-op-text'>

<sup>Originally posted by **tinkerdudeno1** April 27, 2021</sup>

## Describe the bug

Same as the initial bug-report about a year ago, MQTT does not accept any "set" commands, when the "Home Assistant" support has been disabled.

## To Reproduce

1. Get Valetudo up and running, configure mqtt stuff
2. Using an mqtt monitor of your choice (MQTT Explorer works fine) watch the incoming messages at for instance: valetudo/mycoolrobot/FanSpeedControlCapability/intensity when changing the intensity in the webinterface
3. Publish e.g. "high" on valetudo/mycoolrobot/FanSpeedControlCapability/intensity/set and watch the "read"-topic: It will stay on whatever was set on the webinterface. You can see in MQTT Explorer that the message arrived at the broker, but the intensity value is not changing.

This issue is solved, once the "Home Assistant" auto-discover features are enabled. An interesting fact is, that you can then immediately disable the "Home Assistant" features and keeps on working, as long as the topics do exist on the MQTT broker.
So this is definitely an issue on Valetudo, somehow relying on the "Home Assistant" topics on the broker. 
 * If they are not there ("Home Assistant" feature **disabled**), MQTT set **does not** work
 * If they are there ("Home Assistant" feature **enabled**), "MQTT set" **does** work
 * If they are there ("Home Assistant" feature **disabled**), "MQTT set" **does** work

### Vacuum Model

Dreame L10 Pro

### Valetudo Version

2022.08.0

## Expected behavior

Publishing on the set-topic should actually set the value.

systemofapwne, 2022-08-24:
Ok, the thing is even deeper: One more strange thing that I have discovered, right after reporting/"reopening" this issue:

If the MQTT broker (here: Mosquitto without persistent data storage) is restarting, setting a topic **does not** work anymore again.
I have to first stop the MQTT feature on my robot and reenable it. Then the robot pushes all topics back to the broker and then, and only then, it accepts set-commands again.

I think, this is potentially is related to the "Home Assistant" auto-discover feature screwing up "MQTT set", when the "Home Assistant" auto-discover is not set.

Hypfer, 2022-08-25:
> But as soon as I reboot my broker, such that all old topics are gone, the robot will not accept any "/set" commands anymore.

That sounds like it could be the the same issue that was worked around in https://github.com/Hypfer/Valetudo/commit/be33e12148c81353c41dfd7c2d374263efe89a1f

TL;DR:
For some reason, the MQTT library does not resubscribe to topics on reconnect, meaning that for setups with brokers without persistence enabled, Valetudo ended up in a connected state but wasn't listening to any topics at all.

Please try the latest nightly build to see if that commit fixes your issue as well
https://github.com/Hypfer/valetudo-nightly-builds

systemofapwne, 2022-08-25:
I can confirm, that both bugs (Home-Assistent Auto-Discover being necessary and MQTT not working if topics are missing / resubscribing not working, as you described) are fixed in the nightly.

Thank you for clarifying this!

closed 2022-08-25 by systemofapwne