Hypfer/Valetudo #1582: Feature Request - Unattended Valetudo Updade via MQTT / Auto-Update

jwiesel, 2022-09-05:
Hi everyone,
first of all congratulations to making valetudo such a great project and app for cloud-free vacuums!
I bought my roborock after I learned about dustcloud and valetudo some year ago and it's great to see how the project evolved over time!

The embedded updater was a huge step forward for the users to stay up to date! With more and more maturity, it might be possible to go even one step further. It would be great to keep the software up to date, while learning about valetudo improvements.

Idea: Implement an unattended update to the latest valetudo version (either auto-update by valetudo itself or triggered via MQTT message) and show the release notes when opening valetudo the next time.

charliesz, 2022-09-05:
In my opinion that is the wrong path to go. You need to check the changelog before updating, so you can react to different updated features. In my usecase it is not very often, that I use the valetudo gui at all, instead I use my smarthome automation or the smarthome gui. Then there is a chance that maybe my wife gets the changelog instead of me, because she uses the valetudo gui next after an update.

Hypfer, 2022-09-05:
This decision has nothing to do with project maturity.  The updater was purposefully built the way it is now.
It balances user-friendliness and developer-friendliness in a way that makes both parties happy.

I'd like to ask you to be grateful that it even exists at all instead of demanding even more.
The fact that it exists at all is a gift to you as a user. You don't ask people who gift you something for even more gifts.

closed 2022-09-05 by Hypfer

Hypfer, 2022-09-05:
I am also a bit sceptical that right before this attempt to get me to do work that would be your own responsibility, I've received a $10 donation by you so just to be safe:

Friendly reminder that this is not a paid product but done by someone in their free time.

The only reason why you're used to unattended updates with other software is because that you are paying people to take these responsibilities off you. You might only be paying miniscule amounts there, however due to the sheer scale of the operation, it is usually sufficient to sustain that.
Valetudo is not like that. It's laughably small in comparison. Thus, such a support package would likely costs hundreds to thousands of euros per month to make sense financially. Donations aren't payments. Neither in overall amount nor in purpose.

However that is merely theoretical as Valetudo is no business. I am not here to make money.
I am here because I want my vacuum robots to be cloud free and since I've already built everything and sharing is basically free, I might as well just give it to you as well.

jwiesel, 2022-09-05:
Seems like I hit a nerve here.
I've been updating valetudo - which I really like and value - a couple of times on two robots over the past couple of months and just had an idea, I thought I might share. It was not my intention to shift any responsibility or whatsoever.
I now understood that updating valetudo might break things in some cases, and should only be done cautiously.
My payment was based on your request in the last release to support the project. As I've been benefiting from valetudo for some time already, I thought I'd give something back. No relation to my ticket/idea.