Hypfer/Valetudo #1594: partially rotated map

alanpaone, 2022-09-16:
## Describe the bug
Every few weeks, my map gets overlaid with a rotated version, some of the old walls, and all of my virtual walls stay in the same place, but the new ones are rotated about 10 degrees, so the robot might not go down some hallways or it might try to get into my roommates' room. 
I'm not really sure if this is a valetudo issue or a Dreame one, but I've not been able to find anything about it anywhere. it might even be from the angles in my apartment. has anyone seen this happen?

## To Reproduce

Steps to reproduce the behavior:
1. reset the map and create a new one
2. run the robot every day for 1-4 weeks
3. one day it will randomly be stuck somewhere its not supposed to be
4. open the app and the map is now distorted. 

### Screenshots
here's a screenshot of my map
[image]
and here's a floorplan of how the walls actually look when there's only one copy of them. 
[image]

### Vacuum Model
Dreame D9 firmware 1100

### Valetudo Version
22.9.0, but this was happening on 22.8 and more or less all summer

## Expected behavior
hopefully the map would be straight, it at least a consistent angle, and the walls would stay in the same place. 

## Additional context

Hypfer, 2022-09-16:
As Valetudo is not a custom firmware but just a cloud replacement that runs on the robot, it has no influence over how the robot operates. As such, this can't be a bug in Valetudo. Please only open Bug reports for issues where you're 100% sure that it is a bug in Valetudo.

Anyway, you might want to try upgrading the firmware to 1132.
Dreame made a few improvements in that regard.

If that doesn't help then the only option is to see if you can reduce the risk of it getting stuck somewhere.

closed 2022-09-16 by Hypfer

alanpaone, 2022-09-16:
thanks for replying so quickly, I'll try the update

Hypfer, 2022-09-16:
To do that, just build another dustbuilder image and install that via ssh.

If you run into issues after that, make sure to check out this discussion
https://github.com/Hypfer/Valetudo/discussions/1556