Hypfer/Valetudo #1606: Crash on Dreame F9 (2021.05.1)

archi, 2022-10-04:
## Describe the bug

I did not use the F9 for a few weeks [don't judge me! :P], it was just sitting on the dock during that time.
When it did run today, Valetudo crashed a while into the cleaning run.
This never happened before.

The bot itself continued to clean up, but my network status monitoring checks the UI every minute and alarmed me at 12:59 (UTC+2) of the UI being unreachable. The monitoring did not notice a related service having a hickup (mqtt or ntp).

Sadly, [the log file](https://github.com/Hypfer/Valetudo/files/9706394/valetudo.log.txt) looks inconclusive to me, no obvious crash :(
I checked dmesg, but it didn't record anything suspicious (no oom killer or valetudo-related segfault; only their v4l2 kernel module seems to crash occasionally).

I'd be happy to just update Valetudo (as I said, the version is rather old anyway) and have the issue closed, but maybe there is something useful somewhere.

The bot is sitting in the dock again, and I did not yet reboot it; so it can be further investigated.

## To Reproduce
Did not try to reproduce. This is what I did:
1. Start the bot
2. Let it clean for a while
3. Decide to set fan speed to high (which had the expected effect)
4. Let the tab on my phone sit in the background

I then still took a few minutes for the crash to happen, so it's probably not related.

### Screenshots
none

### Vacuum Model
Dreame F9

### Valetudo Version
```
   1 [2022-08-24T12:31:02.021Z] [INFO] Set Logfile to /tmp/valetudo.log
   2 [2022-08-24T12:31:02.052Z] [INFO] Autodetected DreameF9ValetudoRobot
   3 [2022-08-24T12:31:02.234Z] [INFO] Starting Valetudo 2022.05.1
   4 [2022-08-24T12:31:02.237Z] [INFO] Commit ID: a5c8bcdd3394e21b76a76e140fd7a1b57adf0b82
   5 [2022-08-24T12:31:02.238Z] [INFO] Configuration file: /data/valetudo_config.json
   6 [2022-08-24T12:31:02.239Z] [INFO] Logfile: /tmp/valetudo.log
   7 [2022-08-24T12:31:02.241Z] [INFO] Robot: Dreame F9 (DreameF9ValetudoRobot)
   8 [2022-08-24T12:31:02.242Z] [INFO] JS Runtime Version: v18.1.0-Valetudo
```

Date is set via NTP. Seems it was really running that long?

## Expected behavior
No crashing ;-)

## Additional context
I mentioned this on the Telegram channel

Log file, with secrets removed:
[valetudo.log.txt](https://github.com/Hypfer/Valetudo/files/9706394/valetudo.log.txt)

archi, 2022-10-04:
I didn't think too much of these kernel messages, but they seem to only appear at the end of the dmesg ([valetudo-dmesg.log.txt](https://github.com/Hypfer/Valetudo/files/9706498/valetudo-dmesg.log.txt)).

```
[3537440.541696] CPU Budget hotplug: cluster0 min:0 max:4
```

Maybe that's a hint?

Hypfer, 2022-10-04:
Given the warnings thrown by what were likely wifi connectivity issues, I suspect that this was already fixed with this commit

https://github.com/Hypfer/Valetudo/commit/bfb96c09aa2bc0ff9e151fb6cdab4ac5bbd58a36

Please upgrade to the latest version

archi, 2022-10-04:
Hm, unifi says the wifi experience would be good, and the bot is indeed connected to the closest AP:
[image]
The small bump is probably from cleaning the kitchen, which happened before the crash.

Mosquitto log records only the disconnect at 10:52:08 UTC.

But I agree: Updating is the way to go. I think there is no benefit in the issue, since there is no clear indication of what caused the crash; if you agree, feel free to close it :)

closed 2022-10-04 by Hypfer