Hypfer/Valetudo #1623: Updating is impossible because there's not enough space

FarFarFromH0me, 2022-10-28:
## Describe the bug
Just enter the tab to updater and check for updates

## To Reproduce
1. Go to tab Updater
2. Hit button "check for updates"

### Screenshots
[image]

### Vacuum Model
Roborock S7

### Valetudo Version
2022.09.0

## Expected behavior
Update should be small enough to install it

## Additional context
Dont know if its only me or why that happend, always worked in the past

Hypfer, 2022-10-28:
It is unclear to me how a very specific error message reporting not enough disk space could be considered a bug by anyone.

closed 2022-10-28 by Hypfer

Hypfer, 2022-10-28:
Btw, the inbuilt help feature also informs the user about this very issue on some unsupported(!!!!!) robots

[image]

Why are you S7 owners like this