Hypfer/Valetudo #1664: PSA: Where did all those issues and discussions go?

Hypfer, 2022-12-05:
Hi and welcome to the annual maintainer burnout prevention issue.
This time, it's purely informational. Please enjoy this wall of text

## "Where are all those old issues and discussions?"

Gone.

## "Yes, I can see that. But why?"

Because their existence was becoming a burden for me.
Let me explain

First, we need to understand that in western culture, there's one very broken mechanic:
As soon as one participant of a discussion shows an emotional reaction, they set a fail flag that takes precedence over everything else. It doesn't matter what they said, what they've done or where they are coming from or even why there's an emotional reaction.

As soon as there is one, they're out.

This mechanic is also being severely abused by a lot of people trying to "win" by doing everything they can to set this flag and then point at it being set. While I assumed that people would stop being like that as soon as they've left high school, the unfortunate truth is that a lot of people never do leave high school. They don't get older. They just age.

Anyway, since I do care deeply about this project, said flag is practically on by default or at least very close to that, meaning that I'm basically set up to constantly fail with those mechanics in place.

It will come as no surprise that continually entering this failure state _because_ you're passionate about something and _because_ you've given something to the world is taxing.

## GitHub flavoured public bugtrackers

With that out of the way, let's talk about GitHub flavoured public bugtrackers.
I don't like them.

Actually that's an understatement. It's often a strong detractor from the fun that is building Valetudo; so strong that it utterly obliterated my motivation to do anything with the project many times now.

### Why is that?

Well, first of all, what do you do when you receive an email which you don't want to answer because it's a stupid question, someone not reading the documentation or someone once again challenging your design decisions?
You delete it. You're done with it.

You don't have to deal with the emotional fallout of the sender noticing that you did that because they don't receive any feedback.
For all they know, you might not have gotten that email at all. Or maybe you did get it but forgot about it.
They'd likely still be unhappy, but not actively hostile as they tend to not perceive you as the aggressor.

What happens when you receive an issue that you don't want to answer because it's a stupid question, someone not reading the documentation or someone once again challenging your design decisions?

### You delete it?
Ha. That means that now the person on the other end knows that you don't want to deal with that.
This is usually being met with hostility. How dare you silence this person. Someone else could've helped them! If you don't want to help, at least let the community help!!11

### You keep it but don't answer?
Ha. Now you get to watch effectively dead threads staring at you every time that you look at your inbox to see if there is anything new that needs handling.
While 20% of the time, some community member might chip in with an actual helpful reply, most of the time, this issue will now only receive replies such as
- "+1"
- "I also have this issue"
- "I'd love to help" (they never actually want to help)
- "This is still open since X >:("
etc.

Maybe, someone will even come along and suggest something that makes no sense whatsoever. Now you get to watch nonsense unfold in your inbox. Nonsense that you could've prevented because you know the answer, however you did not want to give the author the answer because stupid question, docs, challenging design decisions etc.

Regardless of that, you will eventually give in and answer.
You're not happy with that. You're not happy with the question itself. You think that all this is just stupid.
Congratulations. You have now set the emotional insta-failure flag.
This will be used against you. :-)

**Sidenote:**
Did you ever notice that "issue tracker" greatly extended the scope of what once was a "bug tracker"?
You can't have just a bugtracker on github. It's an issue tracker. You also have to care about support tickets, feature requests etc.

Even though it is your project, you're not allowed to make that decision. The platform dictates that releasing an open source project means that you have to cater to all those things.

## The impact

This quirky implementation of a bugtracker inevitably causes emotional reactions in replies, which, combined with this weird cultural quirk explained above, become failures as far as the public is concerned.
[... 83 lines cut ...]

**Sidenote:**
That's also why there is a "Why not Valetudo?" docs page.
Did you ever see something like that in another project?

I didn't. And they don't have to of course.
The open source license under which we release our software should be enough.

This is the extra mile.

## Final remarks

I'm fairly certain that some readers now feel the strong urge to comment their opinion on this below.
It will most likely include their vision of how things should be done, some suggestions such as "just deal with it", "just ignore it" or my personal favorite "If you don't want to deal with this you should not have released it as open source".

Thank you for your input but not interested.

This is my project which I released under my own terms. I decide how this works.
I don't care how you feel an issue tracker should be used.
I don't care how you think community participation should work.
I don't care about your wrong definition of open source.
I don't care that you think that it's my moral obligation to cater to your or anyone's needs.

If you disagree, simply don't use Valetudo.
No one but yourself is forcing you to use Valetudo.

ballakers, 2022-12-09:
Many thanks for your great work and taking on so much extra work dealing with the public ( including the dicks out there). Keep that chin up and keep turning the other cheek and don’t let stupid comments get to you and keep deleting them! Cheers and again much appreciation for your work!

MonsterEnergy-wtf, 2022-12-11:
Nothing more to add here. 

I also faced the thing that I felt some kind of misunderstood by Hypfer but I fully got his point. And I really respect this.
As I didn't payed this enough I just want to say

THANK YOU FOR YOUR AWESOME WORK

I'm pretty sure that a lot of people out there appreciate your work.
At least I am.

skullydazed, 2023-01-09:
As someone who gave 5 years of his life to a moderately popular open source project all I can say is kudos. I stepped away from my open source project for many of the same reasons you've just explained. You're being extremely generous with your time in releasing Valetudo in the first place, you don't owe anyone anything more.

hacheraw, 2023-01-11:
My cecotec 5090 is not supported and my day is ruined but **all I can say is thank you** for spending hours on this and hope you can handle all the hate (it looks you can anyway)

I'd really love to help but I know nothing about low-level/hacking/firmware related coding so I will patiently wait since I have more faith in open-source than in cecotec programmers. It's been a year since the app stopped working and they just say "hey bro, just push the on/off physical button for a complete clean"

KrzysztofHajdamowicz, 2023-01-12:
> My cecotec 5090 is not supported and my day is ruined but **all I can say is thank you** for spending hours on this and hope you can handle all the hate (it looks you can anyway)
>
> I'd really love to [...]

Yes, it sucks to have obsolete device in deep EOL. As a last resort, you can use Switchbot to push those buttons.

closed 2023-01-21 by Hypfer

bugwelle, 2023-03-22:
Hi,

that's an amazing wall of text.

Thank you very much for writing it, there were many take-aways which were new to me and also valuable. I've got an open-source project myself that I've been maintaining for the last 5 years, and I can feel your pain. And my project only gets about one issue a week. Nothing even close to Valetudo.

While going through your text, I had a few thoughts that I've written down below. Feel free to ignore them, of course. :)

> We should probably talk about expectations for open source software.

I have myself fallen into thinking that projects using GitHub means that I can expect a working issue tracker which is maintained. Only after having to maintain such issue trackers myself, I've changed my mindset.

I now have a "Office Hours / Support" issue pinned in one of my projects that simply states:

1. I have a job, this is a spare-time project
2. Don't expect an answer immedietatly
3. After N days without an answer, you can ping me, but please not any time earlier than that.

That has helped a lot.  Pinned issues stay at the top of the issue tracker, where it's very visible.  I think many users have a look at it, possibly because the title contains "Support".  For me, this kind of expectation management has helped a lot.

You could have a pinned issue that simply refers to Telegram and states that the issue tracker is for bug-reports only, for example.

> What he found was me just saying "no" without any further explanation.

To be honest, I can understand both sides. One the one hand, writing a detailed answer takes _a lot of time_, but on the other hand, just "no" does not tell me anything: "no", because you don't like it? "no", because it's not possible? Or "no", because there is no time?  From a user's perspective it's annoying, from a maintainer's perspective time efficient and good enough.

I've now saved a few standard-responses. One of them says something like:

```
Thank you for this feature request.  Even though it may be useful to you,
the majority of users won't benefit from it and I'm not going to implement it.
```
or
```
Unfortunately, I don't have the time to implement this feature.
See <link to office hours/support issue>.
Pull requests are welcome, though!
```

It may be a bit over the top to always say "Thanks", but this positivity even when declining feature requests has been successful in my opinion. I've had users come back to help test stuff quite a lot.

This may be the "corporate-friendly way" that you refer to. :smile:
But if a user is unfriendly or even threatening, I wouldn't use this corporate-speak.

Of course, there will _always_ be that one person who hates everything and everyone. The screenshot you posted is just horrible.

> Did you ever see something like that in another project?
> I didn't. And they don't have to of course.

And that page is very valuable! I've added an "Alternative Projects" section to a few of my projects, because I often got the question "Why won't you support it? XYZ has it as well!". Yeah, if XYZ has it, then use it? :smile:

Regards,
Andre