Hypfer/Valetudo #1677: Home button reconfigures WiFi on Dreame L10 Pro

RoboMagus, 2023-01-01:
## Describe the bug

When my Dreame L10 Pro stands idle and undocked, short pressing the home button causes it to disconnect from the WiFi and start AP mode, waiting for WiFi reconfiguration.

## To Reproduce

See bug description.

### Screenshots

### Vacuum Model

Dreame L10 Pro (FW version 1138)

### Valetudo Version

2022.11.0

## Expected behavior

Return to the dock

## Additional context

Hypfer, 2023-01-01:
Sounds like the spot cleanup button might be stuck.
That might be caused by the reassembly of that plastic cover.

Please don't open bug reports for things that clearly can't be a bug in Valetudo - a mock cloud running on the robot that does not change anything about the vendor firmware.

closed 2023-01-01 by Hypfer

RoboMagus, 2023-01-02:
> Please don't open bug reports for things that clearly can't be a bug in Valetudo

This isn't something that could be labeled **clearly** unrelated to Valetudo. Seeing as there is the option to disable physical buttons altogether, it is not a far fetched assumption that handling of the buttons could also be altered by Valetudo...

For any outsider this could easily be related to the use of Valetudo.