Hypfer/Valetudo #1700: Segment description does not render in an acceptable scale

cebe, 2023-02-02:
## Describe the bug

Description of Map Segments does not render in normal view, only if I zoom in a lot.

## To Reproduce

1. Open Valetudo
     
[image]

2. Zoom in

[image]

3. only then description appears. 

### Vacuum Model

Roborock S5

RoborockS5ValetudoRobot

Firmware Version 002034

### Valetudo Version

2023.01.0
Commit 3d6ef60758077fc0623474721031c7493e39640d

## Expected behavior

Segment Description should be visible also in normal Zoom level.

Hypfer, 2023-02-03:
Ah yes that might be confusing when coming from a vendor app. That's actually intended behavior to reduce clutter on the map as the information displayed isn't relevant when looking at the full map and likely at a glance

closed 2023-02-03 by Hypfer

cebe, 2023-02-03:
I'm not coming from any vendor app, never had my robot connected to a cloud.
I just upgraded Valetudo from a very old version (something around 2019 :grin: ). You made great progress with it, thanks for that btw!

If it should not appear on the full overview level, it feels like the description could appear a bit earlier (directly after starting to zoom), but I have no strong opinion on it and it might depend on the actual map size.

Hypfer, 2023-02-03:
You can find the logic that is responsible for that behavior here:
https://github.com/Hypfer/Valetudo/blob/2654a3eb98756b44dc643042c639feac9347ed81/frontend/src/map/structures/map_structures/SegmentLabelMapStructure.ts#L70-L123

Feel free to play around with that :)
The current solution certainly isn't the best one