Hypfer/Valetudo #1702: Empty SSID on Viomi SE

rufik, 2023-02-06:
## Describe the bug
WIFI SSID is empty/unknown on Viomi SE.

## To Reproduce
Just convert Viomi SE to Valetudo. Connect to WIFI.

### Screenshots
[image]

### Vacuum Model
Viomi SE

### Valetudo Version
2023.01.0

## Expected behavior
WIFI SSID to be shown.

## Additional context
It's not possible to use special chars (like `#` for me) in wifi password, so I had to create dedicated wifi with simple password. Once vacuum has joined wifi, I could ssh and edit manually `/etc/wifi/wpa_supplicant.conf` file putting special chars in password.

Why special chars in wifi password are forbidden?

Hypfer, 2023-02-06:
Fixed in the next release:
https://github.com/Hypfer/Valetudo/commit/d75514bd5b49a1777a8d4b09554bd6fcde2702e7

closed 2023-02-06 by Hypfer

rufik, 2023-02-06:
👍