Hypfer/Valetudo #1720: Network Advertisement config page shows "Error loading Network Advertisement configuration"

ejohb, 2023-02-24:
## Network Advertisement config page shows "Error loading Network Advertisement configuration"
Network Advertisement config page shows "Error loading Network Advertisement configuration"

## To Reproduce
Visit http://[robot-host]/#/options/connectivity/networkadvertisement

### Screenshots

[image]

### Vacuum Model
This is happening on 2 different S5 Maxes

### Valetudo Version
Both robots running 2023.02.0 

## Expected behavior
Settings page should load

Hypfer, 2023-02-24:
Please check the browser devtools console and network tab for any error messages.
Refreshing the page also often helps

ejohb, 2023-02-26:
The logs implied the request was blocked by something. I disabled my Ad Blocker (Ad Blocker Ultimate) and it started work.

Sorry for the false alarm.

closed 2023-02-26 by ejohb

Hypfer, 2023-02-26:
Sounds like the "advertisement" in the url might've triggered it.

OT, but I can recommend ublock origin as the only trustworthy adblocker
https://github.com/gorhill/uBlock