Hypfer/Valetudo #1740: Security Issue: Request for contact

Hong-Thai, 2023-03-11:
Hallo,
I think found a security issue in this project. It is about the way you encode Miio packages. Can you give me a channel to communicate with you privately, so I can give you more details?

Best,
Hong-Thai

Hypfer, 2023-03-11:
[image]

You mean this? Because that's just the miio protocol.

Hong-Thai, 2023-03-11:
That is a part of the issue. But I think the more critical part is the next line:
`this.tokenIV = crypto.createHash("md5").update(this.tokenKey).update(this.token).digest();`

I'm not sure if this is still part of the protocol, but essentially it holds tokenIV = md5(tokenKey+token). tokenIV is used as an IV for encryption:
`const cipher = crypto.createCipheriv("aes-128-cbc", this.tokenKey, this.tokenIV);`

I'm pretty sure that tokenIV is not supposed to be secret (correct me if I'm wrong). Therefore, one can use tokenIV to retrieve tokenKey+token, since md5 is insecure. Using tokenKey you can than decrypt messages of the user whose tokenIV you've got.

But maybe this is still part of the protocol. What do you think?

Hypfer, 2023-03-11:
This code implements the miio protocol the way the miio protocol works. We could do something else there, but then it would not be the miio protocol anymore and thus we would not be able to talk to devices using the miio protocol with that code.

See for example other documentation and implementations on github:
- https://github.com/OpenMiHome/mihome-binary-protocol/blob/master/doc/PROTOCOL.md#encryption
- https://github.com/rytilahti/python-miio/blob/788df486a5d87008664570972d2f665a0a8705bb/miio/protocol.py#L65-L69
- https://github.com/SG-O/miIO/blob/f352dbd2a699d2cdb1b412ca5e6cbb0c38ca779b/src/main/java/de/sg_o/app/miio/base/Token.java#L86-L89

But even if there was a hypothetical issue in the implementation, it would be completely irrelevant as all that traffic is entirely local on the robot itself. It doesn't even enter the network.
There's just no attack scenario that would not involve already having a shell on the robot and with that there's no point because you already have a shell.

But assuming that there was, what would you do with that?
Decrypt the command traffic between vendor firmware and Valetudo?
Inject own commands?

Okay, fair enough.
In that hypothetical case, you could send a clean command to your own robot on your own network that you could've also sent via the Valetudo REST API.

Epic pwn m8

> What do you think?

I think that you should stop opening unsolicited spam issues based on unvetted findings of an automated code scanning tool.

I also think you should put at least in a bit more effort instead of just pasting the same template to multiple repos.
There's an e-mail address in my profile. If you actually cared about security, you would've easily found that.

This is very likely just part of a greater quest to collect token internet points to pass some arbitrary threshold probably set by your university or similar.

closed 2023-03-11 by Hypfer

Hypfer, 2023-03-11:
Oof okay, I found your uni saarland e-mail address. As expected.

If you've been tasked with doing this - which I assume you have been -, please remind your professor that tasking students with wasting the time of open source maintainers isn't a very nice thing to do.
I doubt that they will care all too much but I wanted to point out that the hostility here is mostly directed at them and not at you since you most likely did not make this decision.