Hypfer/Valetudo #1787: Dreame W10: Carpet sensor setting doesn't disable carpet sensor

AndreKR, 2023-04-16:
## Describe the bug
My Dreame W10 increases the fan speed when it encounters carpet, even though the carpet sensor setting is off:

https://user-images.githubusercontent.com/1188538/232266974-c87e9384-4d55-4648-9a71-350f6026ffa2.mp4

[image]

### Valetudo Version
2023.03.0
fc6b905a99f889473f375eb011bd2d3d2b2599ea

## Expected behavior
I'm assuming with the carpet sensor off, the set fan speed should be maintained regardless of the flooring.

Hypfer, 2023-04-16:
You're looking for the carpet mode setting that can be found in the robot options.

Carpet Mode refers to the increase of the fan speed based on i think brush motor current(?).  The carpet detection sensor quirk instead refers to the Ultrasonic carpet sensor that is used to detect and avoid carpets when the mops are attached.

Confusingly, these two systems are entirely unaware of each other on all supported robots at the time of writing.
They probably did it that way because the carpet mode implementation came first and why change anything about it given that it works.

Anyway, not a bug in Valetudo

closed 2023-04-16 by Hypfer

AndreKR, 2023-04-16:
> You're looking for the carpet mode setting that can be found in the robot options.

And I guess that's not supported on the W10? Because I don't seem to have that setting. My Robot Options look like this:  
[image]