Hypfer/Valetudo #1800: Manual Map Segment Trigger for roborock S6 fails with 'param does not contain any element'

shaymdev, 2023-04-19:
I needed to remap my house, so I deleted the map, used manual control to do a quick drive around the house and the sent the robot home. The map was showing all blue so I tried the quirk "Manual Map Segment Trigger" but it errored with the following (from the log):

QuirksCapabilityRouter: Received error from robot while handling route "/" {
  body: { id: '3e467ac1-7d14-4e66-b09b-8d0554a3194e', value: 'trigger' },
  message: 'param does not contain any element'
}

I dug around and found only two places in code that try to call the command, the one in question:

https://github.com/Hypfer/Valetudo/blob/c261d755bf7e4a84698ec0e1fec70cbaec2cdb33/backend/lib/robots/roborock/RoborockQuirkFactory.js#L218

and this one which includes a `{"map_flag":-1}` parameter:

https://github.com/Hypfer/Valetudo/blob/1d2ad47e9d49eac71f0621d2aa0444acf871acec/backend/lib/robots/roborock/capabilities/RoborockMultiMapMapResetCapability.js#L27

So I tried sending the raw command from the python mirobo. It errored with the following error but the map now shows segments in valetudo. 

`Error: {'code': -10005, 'message': 'save new map too close'}`

I don't know what the `{"map_flag":-1}` does but I'm thinking it (or something similar) needs to be added to the call in the RoborockQuirkFactory.

shaymdev, 2023-04-19:
I created https://github.com/Hypfer/Valetudo/pull/1801 in case that's helpful. I'm new to this so feel free to let me know if I'm out of line with any of this.

Hypfer, 2023-04-21:
Did you use your robot with the vendor app before installing Valetudo? If so, did you do a full factory reset after that?

shaymdev, 2023-04-21:
Yes, I'd used the roborock app with the robot before valetudo but that was 2 or 3 years ago. I'm pretty sure I did a full factory reset (whatever the [root video guide](https://www.youtube.com/watch?v=r_04K5SPEXI&) suggested).

closed 2023-06-27 by github-actions[bot]