Hypfer/Valetudo #1802: Builtin timers run on wrong day when Local<>UTC offset crosses midnight

danderson, 2023-04-22:
## Describe the bug

Setting up full home automation is on my todo, so in the meantime I used Valetudo's builtin timers to set a cleaning schedule.

One of the timers is configured to run on Saturday at 17:00 PDT (my local browser time). However, this timer actually triggers on _Friday_ at 17:00 PDT.

Looking at the saved config JSON, I believe the issue is with the timezone translation. PDT is UTC-7, so when I set the timer to 17:00 local time, the saved config is 17:00+7h = 00:00 UTC. But the days that the timer runs are verbatim what the UI says, so the final config for "Saturday 17:00 PDT" turns into "Saturday 00:00 UTC" in json. Saturday 00:00 UTC is Friday 17:00 local time, so the timer executes 1 day earlier.

Based on this, I think timers end up running 1 day early whenever the requested time + local timezone pushes the UTC time over the midnight line. I guess if you are in India or Australia, and you configure timers early in the morning, it could also happen in reverse, and the robot runs timers 1 day late.

## To Reproduce

1. Set your computer/browser timezone to something America/Vancouver (currently UTC-7)
2. In the valetudo UI, configure a timer to run on Saturday, at 17:00 local time. Save the timer, note that the Valetudo UI shows the translation from 17:00 to 00:00 UTC.
3. Open valetudo_config.json on the robot and examine the timer config. the timer config shows `dow=6` (Saturday)`, `hour=0` and `minute=0`.

### Screenshots

What the saved timer looks like in Valetudo UI, vs. the config saved in json:

[image]

### Vacuum Model

Dreame Z10 Pro, 1156 firmware

### Valetudo Version

2023.03.0 (commit fc73347177709d7b0ff3ad02c3b2e7ae3c024c49)

## Expected behavior

The timer should execute on the day/time set in the UI. Meaning probably, there needs to be even more painful timezone translation logic to account for Local<>UTC being in different days :/

## Additional context

I verified that NTP is enabled on the robot, and it is synchronized to <1s of my other machines.

There is of course an easy workaround once I understood how translation was happening: I just set my timer to run at 16:59, and now it runs on the correct day (Saturday 23:59).

I am willing to go figure out the right fix and send you a PR, if that would be helpful. I know that often reviewing PRs from internet randos is more work than just writing code, so please let me know if you would like me to do that or stay away.

Thank you for Valetudo, it makes me happy every time my robot wakes up and cleans stuff for me :)

Hypfer, 2023-04-22:
Oh! That's a good find indeed. A bug where you can tell that development is mostly done in Europe very close to UTC 🙃 
The same also happens with UTC+2, but it was never noticed as no one scheduled cleanups for 1am. 

I also want to say thank you for the time, effort and respect that went not only into this bug report but also all your communication in general so far! I really appreciate it :)

Fixed with a9228771ac59bd46f4ca0e406915760590aa4297
Thanks for reporting!

<details>
<summary>For reference, here's a spoiler with how it operates now (click me)</summary>

[image]
[image]

[image]
[image]

</details>

closed 2023-04-22 by Hypfer

danderson, 2023-04-22:
Thank you for the change! Timezones always find a way to surprise us developers, no matter what :)

I just deployed HEAD to my robot, and the timer config is indeed storing Sunday 00:00 UTC now. The final proof will be if the timer runs in 5.5 hours from now, but I don't see any reason why not.

Vielen dank!