Hypfer/Valetudo #1847: Discovered MQTT entities with a name that is equal to the device name

Gerben-W, 2023-08-02:
## Describe the bug

After updating Home Assistant to version 2023.8.0 I've got the following warning.

Discovered MQTT entities with a name that is equal to the device name

This stops working in version 2024.2.0. Please address before upgrading.
Some MQTT entities have an entity name equal to the device name. This is not expected. The entity name is set to null as a work-a-round to avoid a duplicate name. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

vacuum.valetudo_robotvacuumcleaner

according tho this link https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/ they made some changes to the MTTQ entries naming.

## To Reproduce

Upgrade Home Assastiant to version 2023.8.0

### Screenshots

### Vacuum Model

Dreame L10s Ultra

### Valetudo Version

2023.05.0

## Expected behavior

## Additional context

N1c093, 2023-08-02:
Here are some additional information from Home Assistant community: https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099

sca075, 2023-08-08:
Actually it is an issue that affects all Home Assistant users. It could be an easy fix but this depends on the time and other functionality the guys here have.

sca075, 2023-08-09:
@Hypfer 
You are correct, anyhow this is to take in account whenever there is the possibility. The early birds catch the fly ;) I mean when you are planning to add new functions or update your excellent UI if this isn't a big deal please consider it on the to do list. For the time be I agree we can set this "warning" to ignore.

shalak, 2023-08-09:
> The main concern about "fixing" this is that fixing it for HA 2023.08 will also mean breaking it for any earlier HA version

I'm not sure I follow - nobody's forcing anyone to update their Valetudos. If they don't want to update HA, they would not update Valetudo and there's no issue then.

Hypfer, 2023-08-20:
9d0fed13cbb6b20ccbb3a0e79111382bcda6b9fc

closed 2023-08-20 by Hypfer