Hypfer/Valetudo #1856: Updater: Error while fetching releases

DerSeb90, 2023-08-14:
## Describe the bug
When I click on "Check for Updates" i get the error "Error while fetching releases"
I'm running Version 2023.05

## To Reproduce

### Screenshots

### Vacuum Model

Dreame L10 Pro
### Valetudo Version

2023.05
## Expected behavior

## Additional context

DerSeb90, 2023-08-14:
Tried an manual Update. It looks like it's an github certificate problem. wget says:
ERROR: cannot verify github.com's certificate, issued by 'CN=DigiCert TLS Hybrid ECC SHA384 2020 CA1,O=DigiCert Inc,C=US':
  Issued certificate not yet valid.
To connect to github.com insecurely, use `--no-check-certificate'.

I think the issue can be closed.

Hypfer, 2023-08-14:
Please make sure that Valetudo is able to reach pool.ntp.org as the robot doesn't have an RTC meaning that each automatic daily reboot by the firmware will cause it to get reset to 1970.

If you do not want Valetudo to access pool.ntp.org, you can also configure a custom NTP server in the settings

closed 2023-08-14 by Hypfer