Hypfer/Valetudo #1972: Shifting maps

spijkers09, 2024-01-23:
## Describe the bug

After a map is created, it is working fine. After a number of weeks (8-12?), the colored part of the map is shifting from the outline. And it appears the robot acts on it. Part of the kitchen section is in the hallway. So when ordered to clean the kitchen, it now starts in the hallway. 
Charging station is glued to its position. Furniture might be shifted slightly, but I would not expect that to influence the map.
## To Reproduce

I have the robot almost a year now, and every 3 months I need to remap the house.
### Screenshots

The first is from 2-3 months ago. The erratic cleaning pattern was due to a dirty sensor.
The second is the reset 2-3 months ago.
The third is the current status.

[image]
[image]
[image]

### Vacuum Model

Dreame D9
### Valetudo Version

2023.10.0 (it was also present in 2023.08.0)
## Expected behavior

I would expect the map to remain as is.
## Additional context

I rooted the robot before the breakout pcb was necessary. The robot works perfectly. The robot normally runs twice a week at night

stefanmd023, 2024-01-23:
The issue is the slippage of wheels, you either have a carpet that is moving or some floor tiles that are too slippery for the robot. Unfortunately this particular Dreame firmware is unable to correct itself. Roborock fixed this issue long time ago.

You can't do much except hope Dreame release a firmware that can correct itself, or removing all slippery areas from the path of the robot.

spijkers09, 2024-01-23:
Ok, thanks for the quick response and explanation. 
My flooring is entirely laminate, I only have 2 doormats, no carpets. But as I am writing this, I can recall the robot getting entangled in some shoe laces recently. That would explain stuff.
Maybe next time I will pick Roborock, but still a happy user!

closed 2024-01-23 by spijkers09