Hypfer/Valetudo #1992: Home Assistant 2024.02 onwards: "The 'schema' option is deprecated" for MQTT auto-discovery

ResteNarquois, 2024-02-03:
Deprecation warnings in Home Assistant logs at startup concerning set up of mqtt.vacuum

In fact, this is not a bug in Valetudo, just a change to how Home Assistant works. It's purely cosmetic, and the vacuums continue to work normally in HA (I have three).  Sample log message:

```
Logger: homeassistant.components.mqtt.vacuum
Source: helpers/config_validation.py:917
Integration: MQTT (documentation, issues)
First occurred: 2 February 2024 at 20:54:42 (3 occurrences)
Last logged: 2 February 2024 at 20:54:42
The 'schema' option is deprecated, please remove it from your configuration
```
Related PR is here:

[https://github.com/home-assistant/core/pull/107274](url)

### To Reproduce
Steps to reproduce the behavior:
1. Start or restart Home Assistant

### Vacuum Model
Roborock S5 (x2)
Dreame L10 Pro

### Valetudo Version
2023.12.0

Hypfer, 2024-02-04:
*sigh\*

_Another_ one of _those_ changes that will have to confuse people with an error message for at least a few weeks, because changing it right away would break the Valetudo integration for all Home Assistant Deployments <= 2024.2 as `legacy` was the default before the change:

https://github.com/home-assistant/home-assistant.io/pull/30666/files#diff-166bc05ce39adccf8e09fdf91a1ac122d1c766fd6f2db7b6e590b41922e3c62aL179-L183

https://github.com/home-assistant/core/commit/9007e17c3e20c76a4446e6cc562daeae960f7293#diff-8adf76ab847b185fa7a1df3dacb4f2713507222a5352c7015c2cc565cb6183b6R53-R56

Sorry people that arrive here from the warning logged by home assistant. I'm aware.
It will be fixed eventually but that will have to wait a few weeks as otherwise, this would force people to upgrade their Home Assistant version right away for no reason.
For now, please just click on ignore.

See also https://github.com/Hypfer/Valetudo/issues/1847#issuecomment-1670819008 where essentially the same thing happened :(

Hypfer, 2024-02-09:
Huh, okay, I just updated my HA instance.
I expected it to raise a warning but it's just a log line that speaking from experience no one reads anyway.

Welp, that was too much worrying for something like that.
Anyway, will be changed eventually but might take a while

closed 2024-02-09 by Hypfer