Hypfer/Valetudo #2056: Roborock S6: GET MapSnapshotCapability kills valetudo

ray0711, 2024-04-28:
### What is happening?

Trying to list the map snapshots to be able to restore later.
Using the inbuilt swagger for an HTTP GET http://XXX.XXX.XXX.XXX/api/v2/robot/capabilities/MapSnapshotCapability
The request get's not response and valetudo is not reachable afterwards.

### Broken or not as expected?

Broken

### What should be happening?

- valetudo stays stable
- valetudo returns an error
- no error should happen but the maps should be listed

### How to reproduce

1. use an roborock S6 with Valetudo 2024.02.0
2. open swagger
3. do an [/api/v2/robot/capabilities/MapSnapshotCapability] Request
4

### Is this Valetudo-related?

Probably Valetudo-related

### Please enter the Valetudo version you're using

 2024.02.0

### Please enter the vendor and model of robot you're using

Roborock S6

### Please enter the robot's firmware version you're using

2652

### Additional context

> [1970-01-01T10:32:50.565Z] [INFO] Set Logfile to /tmp/valetudo.log
[1970-01-01T10:32:50.584Z] [INFO] Autodetected RoborockS6ValetudoRobot
[1970-01-01T10:32:50.723Z] [INFO] Starting Valetudo 2024.02.0
[1970-01-01T10:32:50.726Z] [INFO] Commit ID: 13f3f65fcccd7961e57f0ba9e310ea9e8a23cf65
...
[1970-01-01T10:32:50.734Z] [INFO] Robot: Roborock S6 (RoborockS6ValetudoRobot)
[1970-01-01T10:32:50.734Z] [INFO] JS Runtime Version: v20.10.0-Valetudo
[1970-01-01T10:32:50.735Z] [INFO] Arch: arm
[1970-01-01T10:32:50.738Z] [INFO] Max Heap Size: 42 MiB
[1970-01-01T10:32:50.739Z] [INFO] Node Flags: --expose-gc --max-heap-size=42
[1970-01-01T10:32:50.757Z] [INFO] Autogenerated System ID: UnfortunateTrueRook
...
[1970-01-01T10:32:50.780Z] [INFO] Firmware Version: 2652
[1970-01-01T10:32:50.786Z] [INFO] Current rootfs: system_a (/dev/mmcblk0p8)
[1970-01-01T10:32:51.390Z] [INFO] Valetudo can be reached via: valetudo-unfortunatetruerook.local
[1970-01-01T10:33:04.676Z] [INFO] Starting NetworkConnectionStabilizer
[1970-01-01T10:33:04.687Z] [INFO] Setting /proc/self/oom_score_adj to 666. Previous value: 0
[1970-01-01T10:33:04.689Z] [INFO] Setting process priority to 10. Previous value: 0
[1970-01-01T10:33:04.705Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[1970-01-01T10:33:04.739Z] [INFO] Webserver running on port 80
[1970-01-01T10:33:04.750Z] [INFO] FDSMockServer running on port 8079
...
[2024-04-28T01:40:30.481Z] [INFO] Dummycloud connected
[2024-04-28T07:29:49.992Z] [ERROR] unhandledRejection {
  reason: Error: Received invalid response:unknown_method
      at RoborockMapSnapshotCapability.getSnapshots (/snapshot/Valetudo/backend/lib/robots/roborock/capabilities/RoborockMapSnapshotCapability.js:19:19)
      at async /snapshot/Valetudo/backend/lib/webserver/capabilityRouters/MapSnapshotCapabilityRouter.js:7:22,
  stack: 'Error: Received invalid response:unknown_method\n' +
    '    at RoborockMapSnapshotCapability.getSnapshots (/snapshot/Valetudo/backend/lib/robots/roborock/capabilities/RoborockMapSnapshotCapability.js:19:19)\
n' +
    '    at async /snapshot/Valetudo/backend/lib/webserver/capabilityRouters/MapSnapshotCapabilityRouter.js:7:22',
  promise: Promise {
    <rejected> Error: Received invalid response:unknown_method
        at RoborockMapSnapshotCapability.getSnapshots (/snapshot/Valetudo/backend/lib/robots/roborock/capabilities/RoborockMapSnapshotCapability.js:19:19)
        at async /snapshot/Valetudo/backend/lib/webserver/capabilityRouters/MapSnapshotCapabilityRouter.js:7:22

### Context reflection

nothing

Hypfer, 2024-04-28:
Are you sure about the process crashing thing?
Because that "unhandledRejection" logline is logged by code that handles it. Valetudo should not die from that.

Apart from that, I suppose it's just not supported so I'll remove the capability from the S6 implementation.
It was never really a thing anyway. Roborock quickly abandoned the idea and now it's only the S5 that has it which is also why there is no UI for it

Hypfer, 2024-04-28:
https://github.com/Hypfer/Valetudo/commit/7362514484faa4de86228ce3bff371fff80b9a6c

closed 2024-04-28 by Hypfer

ray0711, 2024-04-28:
You were right. Just retried it, other endpoints still react after calling GET MapSnapshotCapability.
With the Xiaomi App i could restore a previous map, even tough i never figured out how to create multiple snapshots. Still it helped a few times when the robot lost itself completely.

Thanks for the feedback and update :)