Hypfer/Valetudo #2058: Valetudo MQTT disconnecting every 2 minutes

szymucha94, 2024-04-30:
### What is happening?

In short: my backup solution pauses HAOS VM at 1:30 every Monday causing mqtt disconnect from the vacuum robot after which connections last only for 2 minutes until robot is rebooted.
I stayed on one of the 2022 valetudo versions and decided to update to 2024.2 last week. This issue is only visible since yesterday - on old versions mqtt was always perfectly stable.
Using HA 2023.11.2. Robot is Mijia 1C. Attaching screenshots - please let me know what kinds of other logs you need.
Robot is not seen disconnecting from wifi, so this seems like MQTT issue. This is also seen on two identical mijia 1c robots (both running same valetudo versions, both under servers that backup VMs in the same way)

### Broken or not as expected?

Broken

### What should be happening?

MQTT should reconnect and stay connected with no need to reboot the robot.

### How to reproduce

1. Forcibly pause Mosquitto and let the robot disconnect for a moment
2. Resume mosquitto, wait for the valetudo to connect
3. Watch how it reconnects every 2 minutes

### Is this Valetudo-related?

Certainly Valetudo-related

### Please enter the Valetudo version you're using

2024.2

### Please enter the vendor and model of robot you're using

Mijia 1C

### Please enter the robot's firmware version you're using

1122

### Additional context

Issue is seen on two identical robots. This issue did not happen on old (2022) valetudo versions.

### Context reflection

No, it's only the Valetudo version that changed.

### Screenshots

[image]
[image]
[image]
[image]
[image]
[image]

Hypfer, 2024-04-30:
I was indeed able to reproduce the issue.

By pure coincidence, just 3 hours ago(!) there was a new release of the MQTT library used which contains a single bugfix for "keepalive issues": https://github.com/mqttjs/MQTT.js/pull/1855

Please try the latest nightly build and let me know if the issue still persists

bendiger, 2024-05-01:
I'll give it a go and report back! 
Thanks,
~Ben

szymucha94, 2024-05-05:
I switched to "nightly" update branch and attempted update twice. I'm not sure install worked because updater still pops up with the same nightly built.
System information after update:
[image]

szymucha94, 2024-05-05:
Well, I think this update attempt broke my robot, or at least this is a very weird coincidence. When attempting to charge it turns on in lower mode and moves a bit in docking station, they complains about charging error.
Need to reflash this with official build.

Krispkiwi, 2024-05-12:
I noticed something similar which may or may not be related. When I would disable entities in home assistant for the vacuum, I'd check and a few hours later they'd be back. As if the vacuum was routinely dropping off and back onto the instance.

closed 2024-05-20 by Hypfer