Hypfer/Valetudo #2206: Cannot read properties of null (reading 'toString') on WifiConfigurationCapabilityRouter

ruimarinho, 2025-02-09:
### What is happening?

Valetudo crashes on load with error 500 when performing a GET to `/api/v2/robot/capabilities/WifiConfigurationCapability`.

### What should be happening?

The response is returned successfully with status code 200.

### How to reproduce

Just open any Valetudo page as this endpoint is fetched on every refresh.

### Please enter the Valetudo version you're using

2025.01.0

### Please enter the vendor and model of robot you're using

Roborock S5 Max

### Please enter the robot's firmware version you're using

1340

### Additional context

```
[2025-02-09T17:00:00.690Z] [ERROR] Failed auto refresh: TypeError: Cannot read properties of null (reading 'toString')
    at RoborockWifiConfigurationCapability.getWifiStatus (/snapshot/Valetudo/backend/lib/robots/common/linuxCapabilities/LinuxWifiConfigurationCapability.js:39:88)
    at RoborockWifiConfigurationCapability.getWifiStatus (/snapshot/Valetudo/backend/lib/robots/common/miioCapabilities/MiioWifiConfigurationCapability.js:14:26)
    at RoborockWifiConfigurationCapability.getWifiStatus (/snapshot/Valetudo/backend/lib/robots/roborock/capabilities/RoborockWifiConfigurationCapability.js:10:26)
    at WifiConfigurationCapabilityMqttHandle.refresh (/snapshot/Valetudo/backend/lib/mqtt/capabilities/WifiConfigurationCapabilityMqttHandle.js:130:47)
    at RobotMqttHandle.refresh (/snapshot/Valetudo/backend/lib/mqtt/handles/MqttHandle.js:240:25)
    at async RobotMqttHandle.refresh (/snapshot/Valetudo/backend/lib/mqtt/handles/RobotMqttHandle.js:168:9)
```

ruimarinho, 2025-02-09:
Following the stacktrace:

```
[root@rockrobo ~]# iw dev wlan0 link
Connected to xx:xx:xx:xx:xx:xx (on wlan0)
	SSID: xxxxxxxxx
	freq: 2452
	signal: -48 dBm
	tx bitrate: 72.2 MBit/s
```

Hypfer, 2025-02-09:
> Valetudo crashes on load

If Valetudo is still logging things, it is not "crashing".
Please don't use the word crash like that, as it is a very specific kind of fault and cannot be used to describe any fault.
It's easy to mislead people by using words that imply the wrong things

Given that your output of `iw dev wlan0 link` looks correct, did it start working eventually?
Is this a persistent issue or an issue that happened on startup?

ruimarinho, 2025-02-09:
Valetudo presents a loader for about 5 seconds, then renders the page that is being requested and finally starts presenting the loader again. This loop makes the UI unusable as there is not enough time to perform any action before the loader shows up again. Technically, the process is not crashing — that's right — but from a user perspective, it might make future issue search simpler. The exception is being handled, but it does not recover in any way, including a reboot.

Hypfer, 2025-02-09:
> but from a user perspective, it might make future issue search simpler.

Yes and if I wouldn't be vigilant, it would make my life harder by being misled and actually believing that the process crashes.
Precise language is not optional.

Anyway, what confuses me is that to my understanding, the only way `stdout` could be `undefined` would be if the `iw` process did not spawn at all. I can't reproduce that on my S5 Max running FW `1668`.

So maybe firmware?
How did you even manage to run FW `1340` as reported? The oldest firmware available in the dustbuilder is `1662`.

ruimarinho, 2025-02-09:
Let's just say I'm not new to this :) I've just updated my other S5 Max running 1668 and it's experiencing the same issue, so it's unlikely to be the firmware. What I could do is to try and compile a basic binary that just calls `iw`.