Hypfer/Valetudo #2294: Unable to fetch updates OTA

Reinachan, 2025-07-01:
### What is happening?

I'm fetching updates but it fails with this error

```
Error while fetching releases [AxiosError: maxContentLength size of -1 exceeded] 
```

### What should be happening?

It fetches the next update and lets me update

### How to reproduce

1. fetch new updates
2. observe it failing every time

### Please enter the Valetudo version you're using

2024.10, 2024.11, 2025.01

### Please enter the vendor and model of robot you're using

Roborock S7

### Please enter the robot's firmware version you're using

1612

### Additional context

I've done this on several version now, having to manually update from one to the next, starting from 2024.10 and the most recently tested version has been 2025.01

I assume this could be related to an axios bug fixed in 2024 as it seems like the package-lock dependency you're using is from 2022. I'm guessing it's caused by fetching the release page, which now contains lots of very long release notes with images and video as well. But I can't say for sure.
https://github.com/axios/axios/issues/4806

The error message part of the logs (the rest is very long and didn't seem to provide much more information about the issue to my eye. I can gather the rest if wanted though.
```
2025-07-01T01:14:18.067Z error
Error while fetching releases [AxiosError: maxContentLength size of -1 exceeded] {
  code: 'ERR_BAD_RESPONSE',
  config: {
    transitional: {
      silentJSONParsing: true,
      forcedJSONParsing: true,
      clarifyTimeoutError: false
    },
    [...]
```

### Screenshots

[image]

hannesweisbach, 2025-07-04:
Is perhaps your router blocking access to the internet for your robot?

hannesweisbach, 2025-07-08:
> I am also able to wget directly to GitHub from the vacuumer over SSH, so it doesn't make sense for it to not have internet access. I think the issue is elsewhere.

True, I missed that.

Hypfer, 2025-08-31:
Is there anything specific about your setup?
Like perhaps a transparent proxy, low speed internet connection, unreliable ISP, a VPN, you being located in some specific country?

I really tried, but I cannot reproduce this issue locally

Reinachan, 2025-08-31:
The only relevant thing on that list is low internet speed. Or rather, I have 500/500Mbps connection from the router to the ISP and the router handles that fine. But SSH connections to the vacuumer is extremely slow, including transferring files to it. I'm not sure why as other devices don't have this behaviour when connected over 2.4 GHz